mobizt/Firebase-ESP-Client #1: Send_Message examples throw missing header files error (ESP8266)

ChillazZHD, 2021-01-14:
Hey, Im using Arduino Uno to program ESP8266 to send messages to my phone via FCM.
After inserting all keywords i got the following error for both the legacy and HTTPv1 examples:

C:\...\Firebase-ESP-Client-main\src/Firebase_ESP_Client.h:42:17: fatal error: No such file or directory
#include <ETH.h>

I editet the file "Firebase_ESP_Client.h", so that ETH.h will not be imported. The program compiled properly.
I got it working for the HTTPv1 Protocol, but not for legacy, but its possible that there are non related problems.

mobizt, 2021-01-14:
This library is for ESP8266 and ESP32.
You need to install the ESP8266 Core SDK and choose the ESP8266 board.

mobizt, 2021-01-14:
Don't edit the source if you don't know what it is.

mobizt, 2021-01-14:
Ok you are right I accidently edit the source to include ethernet to test but forgot to remove it when publish.

closed 2021-01-14 by mobizt

mobizt, 2021-01-14:
The library updated.
The legacy API, you need to use API Key from Firebase console.

ChillazZHD, 2021-01-14:
Okay thanks.
I have both the server key and device token inserted and it didn't work, however HTTPv1 works fine for me.

mobizt, 2021-01-14:
The legacy FCM works normally. Please check the example. 
You may provide invalid server key which this already stated in the Firebase FCM document.

mobizt, 2021-01-14:
The HTTPv1 uses OAuth2.0 for authenticate while the legacy API use the sever key as in the old 2 years library version and no problem with that.