mobizt/Firebase-ESP-Client #2: Not finding any document

chungalunga, 2021-01-16:
Hi,

I receive 404 not found on Firebase.RTDB.get / set even though I can see the document in the path. Here's what I'm using:

```
#include <Firebase_ESP_Client.h>
FirebaseData fbdo;
Firebase.RTDB.setInt(&fbdo, "document", 1)
```

 Any suggestions?

mobizt, 2021-01-16:
Not found 404 means no request URI allows for that Host.

Your Firebase Host that you provided in the config is not for Firebase RTDB.

closed 2021-01-16 by mobizt

mobizt, 2021-01-17:
You should provide the valid host and all sign in credentials and call the begin function.

https://github.com/mobizt/Firebase-ESP-Client/blob/a5176de94b399dae728b5187db05750ebc8c9ce0/examples/Database/Basic/Basic.ino#L25

https://github.com/mobizt/Firebase-ESP-Client/blob/a5176de94b399dae728b5187db05750ebc8c9ce0/examples/Database/Basic/Basic.ino#L60-L67

chungalunga, 2021-01-17:
Thank you mobitz, I've managed to solve my problem with your help. For any other rookies out there: FIREBASE_HOST is writen in the Data tab of the "Realtime Database" page. Your web api key is found in project settings in the "Web API key" field.