mobizt/Firebase-ESP-Client #4: Unable to initialise via Ethernet

Joelmaca9, 2021-01-22:
**Describe the bug**
Wen using the library with wifi works great, but I am not able to use it trough ethernet 

**To Reproduce**
Steps to reproduce the behavior:
1. connect the ESP32 with ethernet and do Firebase.begin(&config,&auth);

**Expected behavior**
Connect to firebase

**Screenshots**

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40191532  PS      : 0x00060f30  A0      : 0x801957c5  A1      : 0x3f803d30  
0x40191532: Firebase_Signer::handleSignerError(int) at /home/joel/Documents/fw-bird-esp32/VeltiumCharger/components/Arduino/libraries/Firebase-ESP-Client/src/signer/Signer.cpp:516

A2      : 0x3ffc7328  A3      : 0x00000003  A4      : 0x3ffc6460  A5      : 0x00000000  
A6      : 0x3f803bd0  A7      : 0x00000000  A8      : 0x80191532  A9      : 0x3f803d10  
A10     : 0x00000000  A11     : 0x3ffc7f88  A12     : 0xffffffff  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x40098b00  LEND    : 0x40098b0b  LCOUNT  : 0x00000000  

**IDE and its version:**
 - ESP.IDF 4.0

**ESP8266 Arduino Core SDK version**
 - 4.0

mobizt, 2021-01-22:
Even the library does not promote the ethernet support due to lag of SSL/TLS support in ESP8266 until the Core SDK V3.

For ESP32. you should include ETH.h and call ETH.begin();, the connection will be route to ethernet  without modification of WiFiClient code.

This error that you post should not exist if you correctly and successfully using WiFiClient to connect to the server via ethernet.

The library does nothing with this link layer. It should work in Arduino environment using Arduino IDE.

For better decode the stack trace, please use the ESP exception decoder add on in Arduino IDE as the error you post does not address anything related to the ethernet.

closed 2021-01-22 by mobizt

mobizt, 2021-01-22:
You should test your code that related to the Firebase library in Arduino IDE for ethernet, if it's ok, the problem may be in the Ethernet library that use as ESP-IDF component.