mobizt/Firebase-ESP-Client #5: get functions are slow?

star297, 2021-01-25:
Hi,
Any reason why the code snip below takes seconds to process?
On the previous version its pretty much instant.

```
Path = firmwareSubfolder + String("/update");  

if (Firebase.RTDB.getInt(&FBdata, Path.c_str())) {......
```
Many thanks

mobizt, 2021-01-25:
I don't know your auth type and everything about your code.

The time consumed for server connection and data transfer after the token are ready depends on the SSL handshake (few seconds) at the beginning of new http connection.

The http session will be reuse (no SSL handshake), then the consequence data transfer will be faster if you use the FirebaseData object only for normal Firebase RTDB calls i.e. get, set, push, update and delete.

If you use a  FirebaseData object for shared operations between normal Firebase RTDB calls, stream, messaging and storage, the http connection will be close and open for the new connection with SSL handshaking with takes a few seconds to complete.

closed 2021-01-25 by mobizt

mobizt, 2021-01-25:
No previous version of library, if you mean FirebaseESP32 or FirebaseESP8266, it's different.

star297, 2021-01-25:
You shouldn't really close an issue so quickly, there is definitely a problem here.
Sure it could well be me.

I didn't go into too much detail just incase you knew what the cause was.

I have been using your v3.0.7 Firebase ESP8266 Client library (excellent by the way, thank you). 
Same program, I've simply made the migration changes as per your documentation.
So in theory it should work the same?

Same connection set up as per your example:

```
  Serial.printf("Connecting to Firebase\n");

  config.host = FIREBASE_HOST;
  config.api_key = API_KEY;
  auth.user.email = USER_EMAIL;
  auth.user.password = USER_PASSWORD;
  Firebase.begin(&config, &auth);
  Firebase.reconnectWiFi(true);

  Serial.printf("connected\n");
```
This happens very quick, but I'm not sure what your process is.
Do we have a socket re-use connection at this point?
Normally takes longer for me using TLS socket re-use on Mbed.

If so then as you say subsequent data traffic should normally be quick.

I'm only using RTDB set, get and updateNode calls.

But still does not explain why exactly the same program apart from your migration changes has a big impact here.

Many thanks

star297, 2021-01-25:
This is minimum test I did that shows the delay, don't think I've missed anything.

```

#if defined(ESP32)
#include <WiFi.h>
#elif defined(ESP8266)
#include <ESP8266WiFi.h>
#endif
#include <Firebase_ESP_Client.h>

#define WIFI_SSID "SSID"
#define WIFI_PASSWORD "pwd"

#define FIREBASE_HOST "https://project.firebaseio.com"
#define API_KEY       "api-key"
#define USER_EMAIL "email"
#define USER_PASSWORD "pwd"

FirebaseAuth auth;
FirebaseConfig config;

FirebaseData FBdata;

void setup() {

  Serial.begin(115200);
  while (!Serial);

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.printf("\n\nConnecting to Wi-Fi/n");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.printf("\nConnected with IP: ");
  Serial.println(WiFi.localIP());

  Serial.printf("\nConnecting to Firebase\n");

  config.host = FIREBASE_HOST;
  config.api_key = API_KEY;
  auth.user.email = USER_EMAIL;
  auth.user.password = USER_PASSWORD;
  Firebase.begin(&config, &auth);
  Firebase.reconnectWiFi(true);

  Serial.printf("connected\n\n");

  Serial.printf("Send to Firebase....\n");
  if (Firebase.RTDB.setInt(&FBdata, "/LED_Status", 1))
  {
    //Success
    Serial.println("Set int data success");
  } else {
    //Failed?, get the error reason from fbdo
    Serial.print("Error in setInt, ");
    Serial.println(FBdata.errorReason());
  }
  //6. Try to get int data from Firebase
  //The get function returns bool for the status of operation
  //fbdo requires for receiving the data
  Serial.printf("Get from Firebase....\n");
  if (Firebase.RTDB.getInt(&FBdata, "/LED_Status"))
  {
    //Success
    Serial.print("Get int data success, int = ");
    Serial.println(FBdata.intData());

  } else {
    //Failed?, get the error reason from fbdo
    Serial.print("Error in getInt, ");
    Serial.println(FBdata.errorReason());
  }
  
  Serial.printf("\n\nFinished\n");

  while (1) {
    delay(1);
  }
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

mobizt, 2021-01-26:
Please respect my decision and  don't complain about the closing of the issue.

The closing of issue does not mean everything is totally closed or locked, you and anybody can post the comment and I still follow every opened and closed issue.

You said right about this is the issue that caused the huge impact in time consuming of process when migrate to use this library.

At this time I have found some issue in the core library in the function readBytes of WiFiClient class which it's not check the availability of data when the provided read length is larger than the available incoming data. It waits the data to read to be completed as specified length or timed out.

The library [increase the  buffer of read length](https://github.com/mobizt/Firebase-ESP-Client/blob/main/src/database/FB_RTDB.cpp#L3868-L3869) for the response payload which its size is less than 3 bytes, this caused the readBytes function waits for the data to be completed as it will not happen and then timed  (5 seconds). 

This issue only happen in this library and it was fixed already, please update.

Thanks for asking and addressing this issue.

Please read [this](https://github.com/mobizt/Firebase-ESP-Client#authentication) for the time consumed in the data transmission.

star297, 2021-01-26:
That was quick! The update fixed the issue for me :)
Still can't believe how fast that initial SSL socket communication is, just over 1 second.

Many thanks for all your hard work, excellent library!!!

you can consider this issue closed :)  🥇