mobizt/Firebase-ESP-Client #7: token is not ready

imad-123, 2021-01-28:
#include <ESP8266WiFi.h>
//#include <FirebaseESP8266.h>
#include <Firebase_ESP_Client.h>
#define WIFI_SSID "Gal"
#define WIFI_PASSWORD "00000000"
const char* host = "www.google.com"; 
#define FIREBASE_HOST "endoscopiegold.firebaseio.com"

#define API_KEY "AIzaSyCc382c8SlODF-j55zz8lWfsQnSeahV9GQ"
//#define FIREBASE_AUTH "kutF4xktyVbOtxFqjxf1tcJQZH9rXf1A9GomGeIP"
const long utcOffsetInSeconds = 3600;

//Define FirebaseESP8266 data object
FirebaseData fbdo;
FirebaseData fbdo2;
FirebaseData fbdo3;
FirebaseData fbdo4;

FirebaseJson json;
FirebaseJson json2;
FirebaseJson json3;
FirebaseJson json4;

String fireStatus = ""; 
                        
int up = 5; 
int down=4;
int buttondown = 13; // push button is connected
int buttonup = 3;

int tempdown = 0;
int tempup = 0;
int s=0;
// void printResult(FirebaseData &data);
FirebaseAuth auth;
FirebaseConfig config;

void streamCallback(FirebaseStream  data)
{

     if (data.dataType() == "string") {
      Serial.println(data.stringData());
      fireStatus=data.stringData();
      if (fireStatus == "on" ) 
      {        
                                                        // compare the input of led status received from firebase
      Serial.println("Led Turned ON");                                                        
      digitalWrite(up, LOW);                                                         // make external led ON
      } 
      else if (fireStatus == "off") 
      {   
                                                       // compare the input of led status received from firebase
      Serial.println("Led Turned OFF");
      digitalWrite(up, HIGH);                                                         // make external led OFF
    }
    else{
    Serial.println("Command Error! Please send ON/OFF");
    }
    

  } else {
    Serial.println(fbdo4.errorReason());
  }

}

void streamTimeoutCallback(bool timeout)
{
  if(timeout){
    //Stream timeout occurred
    Serial.println("Stream timeout, resume streaming...");
  }  
}
	
void streamCallback2(FirebaseStream  data)
{

    if (data.dataType() == "string") {
      Serial.println(data.stringData());
      fireStatus=data.stringData();
      if (fireStatus == "up" & tempdown == LOW) 
      {                                                          // compare the input of led status received from firebase
      Serial.println("Led Turned ON");                                                        
      digitalWrite(down, LOW);                                                         // make external led ON
      } 
      else if (fireStatus == "stop") 
      {                                                  // compare the input of led status received from firebase
      Serial.println("Led Turned OFF");
      digitalWrite(down, HIGH);                                                         // make external led OFF
    }
    else{
    Serial.println("Command Error! Please send ON/OFF");
    }
    

  } else {
    Serial.println(fbdo2.errorReason());
  }

}

void streamTimeoutCallback2(bool timeout)
{
  if(timeout){
    //Stream timeout occurred
    Serial.println("Stream timeout, resume streaming...");
  }  
}

void setup()
{

  Serial.begin(115200);
  pinMode(up, OUTPUT);
  digitalWrite(up, HIGH);  

  pinMode(down, OUTPUT);
  digitalWrite(down, HIGH); 

  pinMode(buttondown, INPUT_PULLUP);
  pinMode(buttonup, INPUT_PULLUP);
  
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);

  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }

  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

config.host = FIREBASE_HOST;
config.api_key = API_KEY;

  Firebase.begin(&config,&auth);
  Firebase.reconnectWiFi(true);
Firebase.RTDB.setStreamCallback(&fbdo4, streamCallback, streamTimeoutCallback);
 Firebase.RTDB.setStreamCallback(&fbdo2, streamCallback2, streamTimeoutCallback2);

please help me am using platfomio i get this error token is not ready

mobizt, 2021-01-28:
You should follow the examples for authentication.

The [most examples](https://github.com/mobizt/Firebase-ESP-Client/tree/main/examples/Database) except for examples in [Authentication folder](https://github.com/mobizt/Firebase-ESP-Client/tree/main/examples/Authentications), use Email, password and API Key for authentication.

You need to provide the Email and password like [this](https://github.com/mobizt/Firebase-ESP-Client/blob/main/examples/Database/Basic/Basic.ino#L64-L65).

You can't use a lot of Firebase Data object as normal variable.

Firebase Data object  contains the SSL client instance which use large memory when connect to the server.

Your device may be reset due to out of memory. 

**Please don't use more than two Firebase Data objects.**

closed 2021-01-28 by mobizt

imad-123, 2021-01-28:
yes thanks but i get other error password login disabled how to solve that

mobizt, 2021-01-28:
Enable the sign-in provider, see the image below.

[image]