mobizt/Firebase-ESP-Client #8: Support files from SD Card for Firebase.Storage.upload(...) and .download(..) functions

Rafdal, 2021-01-29:
I found that the function `bool FB_CloudStorage::fcs_sendRequest(FirebaseData *fbdo, struct fb_esp_fcs_req_t *req)` inside [src/storage/FCS.cpp](https://github.com/mobizt/Firebase-ESP-Client/blob/main/src/storage/FCS.cpp#L152) doesn't have code for opening files from SD, it's hardcoded for SPIFFS/LittleFS because it doesn't care about the _storageType_ value (of the **req** struct).

That's all :)

Rafdal, 2021-01-29:
If anyone else has this problem, this is a simple workaround:

1. Go to the file src/storage/FCS.cpp of this library.
2. Select lines from [158](https://github.com/mobizt/Firebase-ESP-Client/blob/main/src/storage/FCS.cpp#L158) to [174](https://github.com/mobizt/Firebase-ESP-Client/blob/main/src/storage/FCS.cpp#L174), then delete
3. Paste this:

```
    #warning Using workaround to upload/download from/to SD
    if(req->storageType == mem_storage_type_sd)
    {
        if (!Signer.getCfg()->_int.fb_sd_rdy)
            Signer.getCfg()->_int.fb_sd_rdy = SDFS.begin();
    }
    else if(req->storageType == mem_storage_type_flash)
    {
        if (!Signer.getCfg()->_int.fb_flash_rdy)
            Signer.getCfg()->_int.fb_flash_rdy = FLASH_FS.begin();
    }

    if (req->requestType == fb_esp_fcs_request_type_download)
    {
        #warning Using workaround to upload/download from/to SD
        if(req->storageType == mem_storage_type_sd)
        {
            Signer.getCfg()->_int.fb_file = SD.open(req->localFileName.c_str(), "w");
        }
        else if(req->storageType == mem_storage_type_flash)
        {
            Signer.getCfg()->_int.fb_file = FLASH_FS.open(req->localFileName.c_str(), "w");
        }
    }
    else
    {
        if (req->requestType == fb_esp_fcs_request_type_upload)
        {
            #warning Using workaround to upload/download from/to SD
            if(req->storageType == mem_storage_type_sd)
            {
                Signer.getCfg()->_int.fb_file = SD.open(req->localFileName.c_str(), "r");
            }
            else if(req->storageType == mem_storage_type_flash)
            {
                Signer.getCfg()->_int.fb_file = FLASH_FS.open(req->localFileName.c_str(), "r");
            }
            fbdo->_ss.fcs.fileSize = Signer.getCfg()->_int.fb_file.size();
        }
    }
```

mobizt, 2021-01-29:
Thank you for addressing this issue and contribute the fixes.

Now this issue was fixed in the latest library update which Could Firestore is now supported.

Please update the library.

closed 2021-01-29 by mobizt

Rafdal, 2021-01-29:
That was pretty fast! Thanks ^^

Tested the lastest version of the library, it works like a charm :)

MernSuper1987, 2022-06-04:
Hello Rafdal, Nice to meet you
I am building GPS Tracker application by ESP32
I need to upload gpx file from SD card to firebase storage
Could you please write sample code for everyone?

Best Regards

mobizt, 2022-06-04:
Did you check the examples?

Don't just asking.