mobizt/Firebase-ESP-Client #11: Support for Arduino with Ethernet?

simeon9696, 2021-02-02:
Hi, I really like this package and was wondering if I could simply use it on an arduino with an ethernet shield? Probably swap teh `Wifi.h` library for `Ethernet.h` ?

mobizt, 2021-02-02:
It's not simple to bring Ethernet connectivity with SSL/TLS support in ESP8266, even it is [already included](https://github.com/esp8266/Arduino/pull/6680) in the ESP8266 Arduino repository.

But it's not yet include in the published release v 2.7.4 right now, hope that it will be available in ESP8266 core SDK version  3.x.x.

Some [third party lib](https://github.com/OPEnSLab-OSU/SSLClient) brings SSL/TLS to the embedded device using BearSSL library but it conflicts with ESP8266 with Arduino Core SDK installed as it is already included the compiled version of BearSSL in the Core SDK.

You can't only swap the library because it's different implementation behind.

star297, 2021-02-02:
Same here, if it can be done, this ESP32 device works fine on WiFi but no SSL on Ethernet.
[image]
I have been try all morning, but no success, maybe when the SDK is updated, lets hope.

closed 2021-02-02 by mobizt