mobizt/Firebase-ESP-Client #12: add more than 2 FirebaseData

imad-123, 2021-02-04:
when i use more than 2 FirebaseData it show an error "not connected" please help i need to use more than 2 FirebaseData

mobizt, 2021-02-04:
The Firebase Data object create WiFi (SSL) client instance internally and it used more memory to store the SSL resources.
If you declare the Firebase Data object as global for faster connection (session reused), the free memory available will be lower.

Then you defined  too many objects then you're running out of memory.  The SSL handshaking process at the beginning of erver connection will failed.

This is the limit of memory available in mcu.

You can do as you wish in the limited memory device.

With normal http request (non-secure), you will gain more free memory (http client used less memory) due to no SSL resources reserved.

Firebase server requires only secured connection then you can't use many SSL clients at a time.

closed 2021-02-04 by mobizt

imad-123, 2021-02-04:
ok thanks mr.mobizt