mobizt/Firebase-ESP-Client #13: StoreProhibitedCause Exception using the example Storage/List_Files.ino

Rafdal, 2021-02-05:
**Describe the bug**
I've tried the example [List_Files.ino](https://github.com/mobizt/Firebase-ESP-Client/blob/main/examples/Storage/List_Files/List_Files.ino) with the ESP8266 Wemos D1 Mini Lite but during the request it throws an exception (output at the end).

**To Reproduce**
Steps to reproduce the behavior:
1. Open the example [List_Files.ino](https://github.com/mobizt/Firebase-ESP-Client/blob/main/examples/Storage/List_Files/List_Files.ino)
2. Set all your Wifi and Firebase credentials.
3. Upload and run
4. See error

**IDE and its version:**
 - Platformio Core 5.1.0

**Board and framework info**
PLATFORM: Espressif 8266 (2.6.3) > WeMos D1 mini Lite
PACKAGES: 
 - framework-arduinoespressif8266 3.20704.0 (2.7.4) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)

**Additional information**
- The Storage is working properly, I've used another Firebase.Storage functions like .getMetadata(...) and .upload(...) without problems.
- Only the Wifi and Firebase credentials are the modified parts of the example, the rest of the code is the same
- I just have 23 files on the Firebase Storage, the names are using 8.3 filename convention (from SD)
- I dont know if this might help but according to [a bunch of Serial prints](https://user-images.githubusercontent.com/61258982/107010560-42885e80-6775-11eb-89d3-649acaaecd86.png), the exception could be happening [at line 596 of FCS.cpp (json.get function)](https://github.com/mobizt/Firebase-ESP-Client/blob/main/src/storage/FCS.cpp#L596)

**Serial output**
```
Connecting to Wi-Fi.
Connected with IP: 192.168.1.129

------------------------------------
Get file meta datatest...

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (29):
epc1=0x4000e1b2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffff800 end: 3fffffc0 offset: 0190
3ffff990:  00007b60 00000893 ffffffff 40205e74  
3ffff9a0:  3ffffa90 3ffef490 ffffffff 402082c0  
3ffff9b0:  3ffffa90 40205e74 00000020 40100abe  
3ffff9c0:  00000000 3fff98c8 3ffef490 4022800a  
3ffff9d0:  00000000 3fff74f8 00000000 4020f1ec  
3ffff9e0:  3ffffa0c 00000002 3ffffb90 00000001  
3ffff9f0:  3ffffa90 00000002 3ffffb90 402084e5  
3ffffa00:  3fff74b8 3fff7500 3ffef490 4022800a  
3ffffa10:  00000000 3fff74e4 3ffffe20 40228026  
3ffffa20:  3ffffa48 00000000 3ffffa90 3ffffb24  
3ffffa30:  3ffffbb0 3ffffe20 3ffffa90 402086a4  
3ffffa40:  3ffef490 0000000b 3ffee918 4026b509  
3ffffa50:  00000000 3ffffd70 3ffffe20 4020eecc  
3ffffa60:  3fff749c 3ffffe20 3fff749c 3fff768c  
3ffffa70:  3fff749c 3ffffd70 3ffee918 4020d273  
3ffffa80:  3ffed9d8 3ffeff24 3ffefabc 00000030  
3ffffa90:  00000000 ffffffff ffffffff ffffffff  
3ffffaa0:  00000000 ffffffff 00000000 ffffffff  
3ffffab0:  ffffffff 00000078 00000000 00000100  
3ffffac0:  00000001 3fff6a14 3fff498c 3fff3824  
3ffffad0:  3fff27c4 3fff2cb4 3fff2a8c 3fff2a94  
3ffffae0:  3fff29f4 3fff4a4c 3fff68fc 3fff29ec  
3ffffaf0:  3fff6904 3fff690c 3fff6914 3fff6924  
3ffffb00:  3fff6934 3fff6944 3fff694c 3fff695c  
3ffffb10:  3fff696c 3fff697c 3fff698c 3fff7494  
3ffffb20:  3fff87a8 3ffef490 3ffe8f01 00000008  
3ffffb30:  4025867e 3ffec548 4025862f 3fff752c  
3ffffb40:  3fff753c 3fff753c 00000000 00000000  
3ffffb50:  00000000 00000000 00000000 00000000  
3ffffb60:  00000000 00000000 80000020 00000000  
3ffffb70:  00000000 00000005 00000000 00000000  
3ffffb80:  3ffe8d00 65646e75 656e6966 89000064  
3ffffb90:  00000000 00000000 00000000 00000000  
3ffffba0:  00000000 00000000 00000000 00000000  
3ffffbb0:  3ffef490 40104f7f 3fff2c74 3fff2b44  
3ffffbc0:  3fff768c 3fff74cc 00000005 ffffffff  
3ffffbd0:  ffffffff 00000001 00000000 00000000  
3ffffbe0:  00000001 00000004 ffffffff 00000005  
3ffffbf0:  00000001 00000000 00000001 3fff26c4  
3ffffc00:  3fff3074 3fff281c 3fff252c 3fff2534  
3ffffc10:  3ffeffcc 3ffeffd4 3fff2fc4 3fff2fcc  
3ffffc20:  3fff2fdc 3fff2fbc 3fff2fe4 3fff2fec  
3ffffc30:  3fff2ff4 3fff3004 3fff3014 3fff3024  
3ffffc40:  3fff302c 3fff2c1c 3fff2c2c 3fff2c3c  
3ffffc50:  3fff2c4c 3fff303c 3fff6838 3ffef490  
3ffffc60:  3ffe9660 2c9f0300 4000050c 3fffc278  
3ffffc70:  401025d8 00000000 00000000 00000000  
3ffffc80:  00000000 00000000 00000000 3fff2c5c  
3ffffc90:  3fff2c5c 3fff2c6c 3fff29fc 0021002f  
3ffffca0:  00000056 00000000 00000000 fffffffe  
3ffffcb0:  00000000 00000000 00000000 69727473  
3ffffcc0:  6500676e 86000064 00000103 00000003  
3ffffcd0:  0000002e 0000002e 0000004f 00000003  
3ffffce0:  00000000 00000000 3ffef490 00418937  
3ffffcf0:  3fff405c 3fff305c 00000000 00000000  
3ffffd00:  000000c8 0000087f 0000087f 00000000  
3ffffd10:  00000000 00000000 00000300 00000000  
3ffffd20:  00000000 00000b72 00000000 00000000  
3ffffd30:  00000000 00000000 00000000 00000000  
3ffffd40:  00000000 3ffef490 3fff2c90 3ffef490  
3ffffd50:  3ffef490 3ffef490 3ffef490 3ffef490  
3ffffd60:  3ffef490 3ffef490 3ffef490 40100a87  
3ffffd70:  3fff6800 3fff2a60 80c6a7f0 00000000  
3ffffd80:  00000000 0e13b00b 00000000 00000000  
3ffffd90:  00000000 4bc6a700 00000b88 80000430  
3ffffda0:  00000000 00000000 00000000 00000000  
3ffffdb0:  00000000 00000000 00000000 00000000  
3ffffdc0:  3ffef490 0000005c 3fff2734 0021002f  
3ffffdd0:  09000056 00000000 00000000 40204626  
3ffffde0:  00000000 00000000 3fff0500 69727473  
3ffffdf0:  0000676e 86000b71 3fff0103 00000003  
3ffffe00:  0000002e 0000002e 0000004f 00000003  
3ffffe10:  00000000 00000000 3fff7660 4021c0ea  
3ffffe20:  6f727265 6f632f72 8a006564 3ffef490  
3ffffe30:  ffffffff 3fff36e8 00000000 00000000  
3ffffe40:  00000000 00000d50 00000020 3fff4a68  
3ffffe50:  3fff6c0c 00000000 3fff6024 3fff6024  
3ffffe60:  00000000 0000087f 00000001 0000006e  
3ffffe70:  00000000 3ffeef80 0000087f 00000289  
3ffffe80:  0000087f 00000d7f 3ffeedd4 0000087f  
3ffffe90:  3ffffd70 3ffffd90 00000880 4020dec6  
3ffffea0:  4026b587 3ffffee8 3fffff40 3ffee918  
3ffffeb0:  4026b587 3ffef1f8 3fffff40 4020daa0  
3ffffec0:  3fff2688 000001bb 00000400 4022800a  
3ffffed0:  402206b8 ea1ed9ac 402206b8 40228026  
3ffffee0:  3fff2408 000001bb 3fff4a68 3fff3838  
3ffffef0:  3ffeed18 3ffeef80 3ffee918 3ffe871f  
3fffff00:  3ffeef80 3fffff40 3ffee918 00000000  
3fffff10:  3ffeef80 3fffff40 3ffee918 4020dba8  
3fffff20:  4020e2ec 3ffef230 3ffee918 3ffe871f  
3fffff30:  3ffe875e 3ffeef80 3ffee918 4020dc1b  
3fffff40:  3ffef490 3ffef490 3fff2500 3ffef490  
3fffff50:  00000000 00000005 3ffef230 4020e7cc  
3fffff60:  3ffee7d0 3ffee918 3ffef230 402032c8  
3fffff70:  402206b8 8101a8c0 feefeffe feefeffe  
3fffff80:  feefeffe feefeffe feefeffe feefeffe  
3fffff90:  feefeffe feefeffe feefeffe 3ffef2c8  
3fffffa0:  3fffdad0 00000000 3ffef288 4020f3f4  
3fffffb0:  feefeffe feefeffe 3ffe8538 40100229  
<<<stack<<<

last failed alloc call: 40205E74(2195)

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

mobizt, 2021-02-05:
Your device memory is not enough to handle large JSON payload.
The a lot of JSON meta data returned for each file.
That's it.

closed 2021-02-05 by mobizt

mobizt, 2021-02-05:
In case of large server response payload, you need to increase the BesrSSL rx buffer and response size limit as seen in the setup.