mobizt/Firebase-ESP-Client #14: failed with create/get/patch documents

DanHuang1998, 2021-02-07:
Hello, mobizt, at first thanks you for providing such a complete Firebase lib, and here i test the Firestoreexample in last few days but i get the error.

Then i found the problem is the example code miss the setting config of host and api_key, as the follows. 

1. config.host = FIREBASE_HOST;
2. config.api_key = API_KEY;

But in few days i try to modify the [create_documents example](examples/Firestore/Create_Documents/Create_Documents.ino) that will create a documents 5 min a period, and then it cannot correctly upload a documents to Firestore.
As same as i change the period for 3(or 4) min, and it's works well.
I try to add the httpcode at the failed part, and it got 200 when uploads failed, so i guess the main problem may be that the Firebase connection will lost due to idle too long. So how can avoid this problem?

Here is my test code

  /**
   * Created by K. Suwatchai (Mobizt)
   * 
   * Email: k_suwatchai@hotmail.com
   * 
   * Github: https://github.com/mobizt
   * 
   * Copyright (c) 2021 mobizt
   *
  */
  
  //This example shows how to create a document in a document collection. This operation required Email/password, custom or OAUth2.0 authentication.
  
  
  #if defined(ESP32)
  #include <WiFi.h>
  #elif defined(ESP8266)
  #include <ESP8266WiFi.h>
  #endif
  #include <Firebase_ESP_Client.h>
  
  /* 1. Define the WiFi credentials */
  #define WIFI_SSID "WIFI_AP"
  #define WIFI_PASSWORD "WIFI_PASSWORD"
  
  #define FIREBASE_HOST "PROJECT_ID.firebaseio.com"
  #define API_KEY "API_KEY"
  
  /* 2. Define the project ID */
  #define FIREBASE_PROJECT_ID "PROJECT_ID"
  
  /* 3. Define the user Email and password that alreadey registerd or added in your project */
  #define USER_EMAIL "USER_EMAIL"
  #define USER_PASSWORD "USER_PASSWORD"
  
  //Define Firebase Data object
  FirebaseData fbdo;
  
  FirebaseAuth auth;
  FirebaseConfig config;
  
  unsigned long dataMillis = 0;
  int count = 0;
  
  void setup()
  {
  
      Serial.begin(115200);
  
      WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
      Serial.print("Connecting to Wi-Fi");
      while (WiFi.status() != WL_CONNECTED)
      {
          Serial.print(".");
          delay(300);
      }
      Serial.println();
      Serial.print("Connected with IP: ");
      Serial.println(WiFi.localIP());
      Serial.println();
  
      config.host = FIREBASE_HOST;
      config.api_key = API_KEY;
      /* Assign the user sign in credentials */
      auth.user.email = USER_EMAIL;
      auth.user.password = USER_PASSWORD;
  
      Firebase.begin(&config, &auth);
      Firebase.reconnectWiFi(true);
  
  #if defined(ESP8266)
      //Set the size of WiFi rx/tx buffers in the case where we want to work with large data.
      fbdo.setBSSLBufferSize(1024, 1024);
  #endif
  }
  
  void loop()
  {
  
      if (millis() - dataMillis > 300000 || dataMillis == 0)
      {
          dataMillis = millis();
  
          String content;
          FirebaseJson js;
          
          //We will create the nested document in the parent path "a0/b0/c0
          //a0 is the collection id, b0 is the document id in collection a0 and c0 is the collection id id in the document b0.
          //and d? is the document id in the document collection id c0 which we will create.
          String documentPath = "a0/b0/c0/d" + String(count);
  
          js.set("fields/count/integerValue", String(count).c_str());
          js.set("fields/status/booleanValue", count % 2 == 0);
          js.toString(content);
          
          count++;
  
          Serial.println("------------------------------------");
          Serial.println("Create a document...");
  
          if (Firebase.Firestore.createDocument(&fbdo, FIREBASE_PROJECT_ID, "" /* databaseId can be (default) or empty */, documentPath.c_str(), content.c_str()))
          {
              Serial.println("PASSED");
              Serial.println("------------------------------------");
              Serial.println(fbdo.payload());
              Serial.println("------------------------------------");
              Serial.println();
          }
          else
          {
              Serial.println("FAILED");
              Serial.println("REASON: " + fbdo.errorReason());
              Serial.println("HTTP CODE: " + fbdo.httpCode());
              Serial.println("------------------------------------");
              Serial.println();
          }
      }
  }

mobizt, 2021-02-07:
Thank you for informing this problem.

It's seem to be a bug and will fix and inform you soon.

DanHuang1998, 2021-02-07:
okay, thanks for your reply.

mobizt, 2021-02-17:
The library update is available with many features added and bugs fixed.

closed 2021-02-17 by mobizt

DanHuang1998, 2021-02-25:
Thanks @mobizt .It's works well after latest update.
But here i facing some problem. At few days i try my prior Firebase Stream code. It's got the exception and the exception is will caused randomly.
In my code i use two FirebaseData object and a FirebaseJson object. An FirebaseData object which i called fbdo1 is used to streaming and the streaming data will stored to the FirebaseJson object. Another FirebaseData object i called fbdo2 is used to get and post data to database.
At the setup, i used both FirebaseData objects to write some data to database, but after the loop, i only use fbdo1 to receive streaming data and fbdo2 use to write and recieve.
But let me confused is, the code will be crashed at sometime. So l try to deploy the code in two different environment. The first environment(env1) which wifi signal is strong, and another environment(env2) which signal is very weak, may connection lost or recieve data slowly(maybe is the main problem). At the env1, the code is works well, and now i'm deploying it. But the env2, it may caused wdt reset at sometime. And it's amazing is, the caused exception code is not the same, maybe caused exception29, exception9, or another situation.
So i used espExceptionDecoder to decode it, and it get the follows error. I'm guessing it may caused by the environmentwhich due to the WIFI signal is weak, so when it handle request, it can't completely receive, or get the mistake.

DanHuang1998, 2021-02-25:
Here is my first exception code:

Exception (9):
epc1=0x40257e1e epc2=0x00000000 epc3=0x00000000 excvaddr=0x0c0b094f depc=0x00000000

>>>stack>>>

ctx: bearssl
sp: 3fff31c8 end: 3fff3638 offset: 0190
3fff3358:  00000057 3fff8c48 00000999 40257769  
3fff3368:  f7e4f31c 7d72384c 9097804f dc879ea2  
3fff3378:  4bd90a3c 46c91510 037a7760 234e6040  
3fff3388:  d19b5393 4ca06a60 a26ad66d 64a4e2f5  
3fff3398:  f9211896 51667fc3 757cb534 c97c51ba  
3fff33a8:  96747ab7 4a51d3de 1735531d 7167392e  
3fff33b8:  8013df2a 854583ff 95b6014d 9b30d5eb  
3fff33c8:  7459c77e c5beb60b c1ad9dff c37e197a  
3fff33d8:  f9211895 e546a8fd 5d5bd821 d272a8bd  
3fff33e8:  cbc52b12 c1839111 6dc47613 881d8a55  
3fff33f8:  5d418c15 86e1b455 e0bd386e a126c97b  
3fff3408:  6c1fd6c6 1820dd13 f709a8fd 3fff34b4  
3fff3418:  6862ddea 96747ab7 d272a8bd 4a51d3de  
3fff3428:  00000000 e0a8a779 00000000 00000002  
3fff3438:  3fff5e6f c4f1d709 1735531d 4025c445  
3fff3448:  0000276f 00005500 00008435 000000a0  
3fff3458:  3fff35e8 000000ad 00000077 4025c624  
3fff3468:  000017c0 0000780f 0000034d 00000253  
3fff3478:  00001c3e 000087cd 00001b85 00002de6  
3fff3488:  0000021c 000000ab 000004c0 00001803  
3fff3498:  000000a9 00000077 000005a6 00001b29  
3fff34a8:  00000581 0000092e 0000006c 00001e3e  
3fff34b8:  00000ecb 000018b5 00000e99 00001d5c  
3fff34c8:  00000c8d 00001e71 0000069e 00001114  
3fff34d8:  00001299 fc9800ab 377054e0 7eca4b4f  
3fff34e8:  dd892e2c 6e78224a a23260b6 a68d74da  
3fff34f8:  7f1dd4ac 0000000d 00000000 0000001e  
3fff3508:  00000000 3fff4e61 000000a2 3fff3558  
3fff3518:  00000044 0000001e 00000070 3fff682c  
3fff3528:  3fff3585 3fff5e51 0000002c 3fff3578  
3fff3538:  000000dd 00000000 00000000 40258b00  
3fff3548:  3fff5261 4025b7ec 00000001 00000000  
3fff3558:  00000000 00000000 00000000 40258b00  
3fff3568:  3fff35e8 4025b7ec 00000000 00000030  
3fff3578:  00000000 0b000000 00030317 1fd6c61e  
3fff3588:  20dd136c 4028d7db deadbef7 deadbeef  
3fff3598:  3fff8968 3fff89ec 00000003 3fff8948  
3fff35a8:  0000001e 3fff8164 3fff8968 40258bc5  
3fff35b8:  00000001 00000057 3fff8148 00000000  
3fff35c8:  3fff3618 0000001e 3fff5e51 3fff8b64  
3fff35d8:  3fff8c64 00000000 3fff8164 40255e18  
3fff35e8:  3fff8948 00000540 9e29cdf6 d24ee8e1  
3fff35f8:  00000450 00000005 deadbeef 3fff0160  
3fff3608:  00000000 00000307 3fff8164 402562fd  
3fff3618:  000000f0 deadbeef deadbeef 402560ec  
3fff3628:  deadbeef 3fff1d34 00000004 402336b4  

ctx: cont
sp: 3ffffbb0 end: 3fffffc0 offset: 0000
3ffffbb0:  00000000 3fff1d34 00000000 40235c25  
3ffffbc0:  00003a98 01a3be1b 3fff4c00 00000540  
3ffffbd0:  00000010 4bc6a7f0 e6a7ef9d 00000000  
3ffffbe0:  00000006 4bc6a7f0 f0e56041 3ffffe90  
3ffffbf0:  3fff4380 3ffffe90 3fff1d34 40235dcd  
3ffffc00:  4027ddfa 3ffef5f0 3fff1d34 40235494  
3ffffc10:  4027efe7 3ffef5f0 3ffef5f0 40239d32  
3ffffc20:  3fff08bc 40206804 00000000 4022453c  
3ffffc30:  3ffef5f0 3ffef5f0 3ffef5f0 4021b01d  
3ffffc40:  3fff4380 3ffffe90 3ffef5f0 402143d7  
3ffffc50:  3ffffde8 3fff0678 00000000 3ffffe90  
3ffffc60:  3ffef5f0 3ffef990 3ffe8cd0 402156e4  
[... 88 lines cut ...]
0x401008c3: umm_free_core(void*) at C:\Users\as098\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\umm_malloc\umm_malloc.cpp line 351
0x40100bc3: malloc(size_t) at C:\Users\as098\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\umm_malloc\umm_malloc.cpp line 552
0x40212fa1: std::_Function_base::_Base_manager    (FB_RTDB*)> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at c:\users\as098\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2/functional line 1910
0x40226894: operator new(unsigned int) at C:\Users\as098\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\abi.cpp line 39
0x40212fa1: std::_Function_base::_Base_manager    (FB_RTDB*)> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at c:\users\as098\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2/functional line 1910
0x40206344: std::function ::function(std::function  const&) at c:\users\as098\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2/functional line 2437
0x401001b5: schedule_function(std::function  const&) at c:\users\as098\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits/move.h line 175
0x40100b88: free(void*) at C:\Users\as098\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\umm_malloc\umm_malloc.cpp line 398
0x4020a71d: std::_Function_handler  )::{lambda()#1}>::_M_invoke(std::_Any_data const&) at c:\users\as098\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2/functional line 2073
0x40246cb8: std::basic_string  , std::allocator  >::basic_string(std::string const&) at /workdir/arena.x86_64/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc line 176
0x40100366: millis() at C:\Users\as098\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\core_esp8266_wiring.cpp line 188
0x40221fd9: Firebase_Signer::getToken(fb_esp_auth_token_type) at .pio\libdeps\nodemcuv2\Firebase Arduino Client Library for ESP8266 and ESP32\src\signer\Signer.cpp line 1793
0x4021b640: FirebaseData::validRequest(std::string const&) at .pio\libdeps\nodemcuv2\Firebase Arduino Client Library for ESP8266 and ESP32\src\session\FB_Session.cpp line 666
[... cut ...]
0x40100b88: free(void*) at C:\Users\as098\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\umm_malloc\umm_malloc.cpp line 398
0x40245b19: operator delete(void*) at /workdir/repo/gcc/libstdc++-v3/libsupc++/del_op.cc line 48
0x40100366: millis() at C:\Users\as098\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\core_esp8266_wiring.cpp line 188
0x40203c80: loop() at src\main.cpp line 69
0x40226b87: loop_wrapper() at C:\Users\as098\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\core_esp8266_main.cpp line 197

DanHuang1998, 2021-02-25:
And here is the seconde exception.

Exception (29):
epc1=0x40246809 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffff9c0 end: 3fffffc0 offset: 0190
3ffffb50:  00000001 000003e9 000007bc 40246804  
3ffffb60:  000003b1 00000020 3fff8c54 40246a6d  
3ffffb70:  3fff60fc 000003e9 3ffffe10 40246ac4  
3ffffb80:  00000001 000003de 000003de 4024681c  
3ffffb90:  40289da0 000003e9 3ffffe10 40246b8d  
3ffffba0:  3fff8c60 00000020 00000010 40100bc3  
3ffffbb0:  00000001 40208f76 00000010 3ffffe10  
3ffffbc0:  40289cd0 00000010 3fff60fc 40208fb0  
3ffffbd0:  00000010 4bc6a7f0 60c49ba5 3ffefbb0  
3ffffbe0:  00000001 3ffffe90 3ffffe10 40214f56  
3ffffbf0:  00000000 00000023 3ffef5f8 4024696d  
3ffffc00:  4027ddfa 3ffef5f0 3fff1d34 3fff4a28  
3ffffc10:  3fff4380 00000b3b 00000b3b 401008c3  
3ffffc20:  c02899f3 00000020 0000000a 3ffffe90  
3ffffc30:  3ffef5f0 00000020 3fff615c 40100b88  
3ffffc40:  00000003 00000000 00000000 40245b19  
3ffffc50:  3fff615c 00000000 00000000 3ffffe90  
3ffffc60:  3ffef5f0 3fff615c 00000000 40215894  
3ffffc70:  3fff4858 3fff9438 3fff066c 40250a48  
3ffffc80:  3fff08bc 40206804 3fff1d34 40235494  
3ffffc90:  00000014 3fff9434 000000ff 00000000  
3ffffca0:  00000006 3fff0604 00000000 4022453c  
3ffffcb0:  3fff9434 00000020 00000010 3fff9434  
3ffffcc0:  3fff4474 3fff066c 00000000 40250a90  
3ffffcd0:  3fff066c 00000000 00000000 40247ad1  
3ffffce0:  3fff0288 0000005a 00000000 40250ab6  
3ffffcf0:  00000001 4bc6a7f0 34bc6a7f 00000000  
3ffffd00:  00000001 00000000 4bc6a7f0 40246cb8  
3ffffd10:  3fff066c 00000a50 00000a50 401008c3  
3ffffd20:  00000001 4bc6a7f0 24dd2f1a 00000000  
3ffffd30:  00000001 00000000 4bc6a7f0 00000000  
3ffffd40:  3fff5a04 0000002f 0000000c 40100bc3  
3ffffd50:  00000001 40212fa1 0000000c 40226894  
3ffffd60:  0dc1abf2 3fff68e4 3ffffd90 40212fa1  
3ffffd70:  00000000 3fff415c 3ffffd90 40206344  
3ffffd80:  00000001 3fff415c 3fff59c4 401001b5  
3ffffd90:  3fff68e4 00000020 00000000 00000000  
3ffffda0:  3fff5aa4 00000020 00000020 40100b88  
3ffffdb0:  3ffeffe0 3fff415c 3fff4164 4020a71d  
3ffffdc0:  00000001 4bc6a7f0 3c6a7ef9 00000000  
3ffffdd0:  00000001 00000000 4bc6a7f0 40246cb8  
3ffffde0:  3fff60fc 3ffffe90 40100366 000be5c6  
3ffffdf0:  00000001 00000020 3ffffe2c 40221fd9  
3ffffe00:  0dc1ad35 3ffef5f0 00000000 00000001  
3ffffe10:  3fff8c60 3fff0288 3ffef5f0 4021b640  
3ffffe20:  3ffefbb0 3ffef5f0 3ffeffe0 3fff4a28  
3ffffe30:  3ffef5f0 0000001e 3ffeffe0 40223cf1  
3ffffe40:  3ffefbb0 3ffffeac 3ffef5f0 3ffefbb0  
3ffffe50:  3ffffe90 00000000 3ffef5f0 40218dd0  
3ffffe60:  3ffef9f0 3fff027c 3ffffe90 40213e8d  
3ffffe70:  3fff0288 0000135b 0000135b 3fff00b4  
3ffffe80:  3ffefbb0 3ffef5f0 3fff9ff4 4021910a  
3ffffe90:  3fff9c98 00000003 00000007 3fff0288  
3ffffea0:  3fff0288 3fff0288 3ffffe00 3fff0288  
3ffffeb0:  00000000 00000000 3fff0018 40225be8  
3ffffec0:  3fffdad0 00000000 3fff0018 402039a8  
3ffffed0:  4021863c 00000000 80000001 3fff5900  
3ffffee0:  0015001f 80ffda00 3fff00b4 00000030  
3ffffef0:  00000000 3fffa200 001e001f 801008c3  
3fffff00:  00000000 3fffdad0 3fff00b4 3fff00b4  
3fffff10:  00000001 0000002f 3fff615c 40100b88  
[... 42 lines cut ...]
0x401008c3: umm_free_core(void*) at C:\Users\as098\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\umm_malloc\umm_malloc.cpp line 351
0x40100bc3: malloc(size_t) at C:\Users\as098\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\umm_malloc\umm_malloc.cpp line 552
0x40212fa1: std::_Function_base::_Base_manager    (FB_RTDB*)> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at c:\users\as098\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2/functional line 1910
0x40226894: operator new(unsigned int) at C:\Users\as098\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\abi.cpp line 39
0x40212fa1: std::_Function_base::_Base_manager    (FB_RTDB*)> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at c:\users\as098\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2/functional line 1910
0x40206344: std::function ::function(std::function  const&) at c:\users\as098\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2/functional line 2437
0x401001b5: schedule_function(std::function  const&) at c:\users\as098\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits/move.h line 175
0x40100b88: free(void*) at C:\Users\as098\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\umm_malloc\umm_malloc.cpp line 398
0x4020a71d: std::_Function_handler  )::{lambda()#1}>::_M_invoke(std::_Any_data const&) at c:\users\as098\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2/functional line 2073
0x40246cb8: std::basic_string  , std::allocator  >::basic_string(std::string const&) at /workdir/arena.x86_64/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc line 176
0x40100366: millis() at C:\Users\as098\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\core_esp8266_wiring.cpp line 188
0x40221fd9: Firebase_Signer::getToken(fb_esp_auth_token_type) at .pio\libdeps\nodemcuv2\Firebase Arduino Client Library for ESP8266 and ESP32\src\signer\Signer.cpp line 1793
[... cut ...]
0x40100b88: free(void*) at C:\Users\as098\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\umm_malloc\umm_malloc.cpp line 398
0x40245b19: operator delete(void*) at /workdir/repo/gcc/libstdc++-v3/libsupc++/del_op.cc line 48
0x40100366: millis() at C:\Users\as098\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\core_esp8266_wiring.cpp line 188
0x40203c80: loop() at src\main.cpp line 69
0x40226b87: loop_wrapper() at C:\Users\as098\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\core_esp8266_main.cpp line 197

mobizt, 2021-02-25:
The first decode error is you don't have enough memory for SSL handshaking.
The second decode error is you don't have enough memory for making the http request header string.

Both errors occurred when you call getJSON.

You free memory is not enough for making a request.

DanHuang1998, 2021-02-27:
Okay. I test my code in last two days. If i only use streaming to a RTDB node(call node1), the code seems works well, and same test result to only receive another node(call node2) periodically, but wdt reset will triggerd if use streaming to node1 and receive node2 periodically.

I also use the esp8266 api to get free heap size(`system_get_free_heap_size()`), and i observe that if it receive data successfully, `system_get_free_heap_size()` will return almost 20500 bytes, then if it receive data failed, and the `FirebaseData.errorReason()` return "not connect", `system_get_free_heap_size()` will return about 15000 bytes, or lower it. Contrary, if the `FirebaseData.errorReason()` return connection lost, the return value will higher than successfull situation(about 27000 bytes).

In my whole code i clear the FirebaseData object everytime i received data(only clear the receive FirebaseData object, the streaming FirebaseData object will not do this), and cleared FirebaseJson object, and deconstructor the FirebaseJsonData object at the end of the function if i used them.

Could i effectively reduce used heap size by doing this? Or is it unnecessary?

mobizt, 2021-02-27:
The free memory required for SSL client operation especially in handshaking and will not free until the connection was closed. The SSL handshaking may fail when no enough memory.

Call the member function clear from Firebase Data object also stop the SSL client and release the resources.

Call the member function **clear** from FirebaseJson will clear the  internal dynamic allocation memory but some are not freed until the object goes out side the scope. The FirebaseJsonData class doesn't have the clear function. 

The memory also reserved by internal SSL client for  buffering  with function **setBSSLBufferSize** of the Firebase Data object.

The stream Firebase Data object should be defined as global while Firebase Data object  and  FirebaseJson object can define as local var.

You should test everything and manage your own usages which I can't recommend or give the solution based on your situation.

DanHuang1998, 2021-03-01:
Okay. I will continue to test. Thanks for your clearly explanation, i will post the main problem here if i found it.

DanHuang1998, 2021-03-04:
Hello @mobizt, in this few days i try several experiment with my code and example code.
In my code i try to
1. only stream to a parent node (which have 5 children node)
2. stream to parent node and get other children node (called node2, which have 12 children nodes)
3. only get children node (node2)

Then the test result still crashed at sometimes, and before the code crashed, the output will occur that like not connected or read Timeout.
So after this, i also test the [example ](https://github.com/mobizt/Firebase-ESP-Client/blob/704a2959a649ffe2635ae05615d66c3372af9fb1/examples/RTDB/MultiPath_Stream/MultiPath_Stream.ino).

In example code i also got crashes at some time, but before the crashes, it seems like have the same situation with my code.
In my guess the crashes may be caused by poor WIFI signal, which caused the memory unable to release normally.
Here is the example code's whole of output and exception:
[opt1.txt](https://github.com/mobizt/Firebase-ESP-Client/files/6080337/opt1.txt)
[exception decoder output.txt](https://github.com/mobizt/Firebase-ESP-Client/files/6080356/exception.decoder.output.txt)

mobizt, 2021-03-04:
If you want to get the actual stream event data payload, in this case for testing, don't use multiPathStream which do post processing the data and can be failed in truncated, incomplete JSON, use normal stream functions.

The crash can be in the id token exchange and refreshment process which required time and resources to handle the token.

If you want stable operation but less secure at this time, use the legacy token (database secret).