mobizt/Firebase-ESP-Client #16: stream slow with autoconnect

imad-123, 2021-02-08:
please help me stream become slow when i use autoconnect library

mobizt, 2021-02-08:
Thanks for inform this problem.

This is the timeout and session bugs which widely affected to [other classes](https://github.com/mobizt/Firebase-ESP-Client/issues/14) too.

I'm currently prepare the update with new features and this bugs fixes. 

The update will be available soon next few days.

I'll inform you when available.

mobizt, 2021-02-08:
Don't use other autoconnection libraries as this library handle this internally and may interrupted in operation to each others.

imad-123, 2021-02-08:
thanks for replay mobizt i need to use autoconnect or wifimanager to change ssid and pwd remotly but in both libraries firebase streaming become slow

mobizt, 2021-02-08:
You should give your task the priority.
When wifi ready and your most priority is Firebase you should disable other reconnect in others libraries.
When wifi is not ready you can set false to [this function](https://github.com/mobizt/Firebase-ESP-Client/blob/main/examples/RTDB/Basic/Basic.ino#L68) to allow other libraries control the wifi.

imad-123, 2021-02-10:
yes thanks mobizt it works but i have a problem in the second FirebaseData it gives not connected error

mobizt, 2021-02-10:
You should check the internet connection in your code yourself because you're not allow the library to handle this and the error "not connect" is the only show when can't connect to server.

mobizt, 2021-02-17:
The library update is available.

closed 2021-02-17 by mobizt