mobizt/Firebase-ESP-Client #17: HELP

Niltonsf, 2021-02-09:
@mobizt 
Hello, I'm having a problem when using a function from the Firestore configurations. When I attempt to use ```Firebase.begin(&config, &auth);``` inside of my function, I get the following error and the device restarts:
Error:
```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400dda5b  PS      : 0x00060e30  A0      : 0x800df942  A1      : 0x3ffd0420  
A2      : 0x3ffc7290  A3      : 0x00000002  A4      : 0x3ffc7700  A5      : 0x3ffc59e8  
A6      : 0x3ffe9724  A7      : 0x3fff65ec  A8      : 0x800dda5b  A9      : 0x3ffd0400  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x000000ff  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x400dda5b:0x3ffd0420 0x400df93f:0x3ffd0460 0x400e0e2d:0x3ffd0570 0x400d6310:0x3ffd0590 0x400d275b:0x3ffd0640 0x400d31c9:0x3ffd0860 0x400d20fd:0x3ffd09a0 0x400e4b31:0x3ffd09c0 0x4008e8e9:0x3ffd09e0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
```
The code, this is a cpp file:
```

// FIRESTORE CONFIG _________________________________________________________________________________________________________
#define FIREBASE_HOST "automacao-com-aplicativo.firebaseio.com"
#define FIREBASE_PROJECT_ID "automacao-com-aplicativo"
#define API_KEY "AIzaSyDErYVplAvwdZ2LpoKFNRzc4Olnp0DMHaY"

//Define Firebase Data object
FirebaseData fbdo;

//Json object to access data
FirebaseJsonData jsonData;

FirebaseAuth auth;
FirebaseConfig config;

// FIRESTORE CONFIG _________________________________________________________________________________________________________

void funcClass::create_document_database(char email[], char password[]){
  int count = 0;

  delay(1000);
  
  SerialBT.disconnect();

  Serial.println("Connecting to WiFi");
  WiFi.mode(WIFI_STA);
  WiFi.begin("test", "test");

  unsigned long startAttemptTime = millis();

  while (WiFi.status() != WL_CONNECTED && millis() - startAttemptTime < WIFI_TIMEOUT_MS)
  {
    Serial.print(".");
    delay(500);
  }

  if (WiFi.status() != WL_CONNECTED){
    Serial.println("\nConnection FAILED!");
    delete_file_info();
    return;
  }else{
    Serial.println("\nSuccessful Connection!");
  }

  config.host = FIREBASE_HOST;
  config.api_key = API_KEY;

  auth.user.email = "test@test.com.br";
  auth.user.password = "test";

  Firebase.begin(&config, &auth);
  Firebase.reconnectWiFi(true);

  String content;
  FirebaseJson js;
  
  //We will create the nested document in the parent path "a0/b0/c0
  //a0 is the collection id, b0 is the document id in collection a0 and c0 is the collection id id in the document b0.
  //and d? is the document id in the document collection id c0 which we will create.
  String documentPath = "a0/b0/c0/d" + String(count);

  js.set("fields/count/integerValue", String(count).c_str());
  js.set("fields/status/booleanValue", count % 2 == 0);
  js.toString(content);
  
  count++;

  Serial.println("------------------------------------");
  Serial.println("Create a document...");

  if (Firebase.Firestore.createDocument(&fbdo, FIREBASE_PROJECT_ID, "" /* databaseId can be (default) or empty */, documentPath.c_str(), content.c_str()))
  {
      Serial.println("PASSED");
      Serial.println("------------------------------------");
      Serial.println(fbdo.payload());
      Serial.println("------------------------------------");
      Serial.println();
  }
  else
  {
      Serial.println("FAILED");
      Serial.println("REASON: " + fbdo.errorReason());
      Serial.println("------------------------------------");
      Serial.println();
  }
}

mobizt, 2021-02-09:
Please be careful when using the library.

You **must call** Firebase.begin(&config, &auth) in setup to initialize the library.

If you  put this in your class function, it should call first and should no return statement  before the initialization as in your code like this.

Any Firebase call  in your other functions without begin initialization will lead to error and the config passed to the library as a pointer to the config data. 

If pointer points to nothing then the error will be occurred.

```cpp
 if (WiFi.status() != WL_CONNECTED){
    Serial.println("\nConnection FAILED!");
    delete_file_info();
    return; //<-------------------
  }else{
    Serial.println("\nSuccessful Connection!");
  }
```

closed 2021-02-09 by mobizt

Niltonsf, 2021-02-09:
@mobizt 
So I should call the function before initialization WiFi? 
I receive parameters from the function, in this case the ssid and the password, than I call the Firebase.begin()

mobizt, 2021-02-09:
I never tell you that it should call before  WiFi initialization.

Did you think this library work for something without internet?

mobizt, 2021-02-09:
The library initialization is the most important process. 

It do a lot of works behind which requires the resources and time to process.

It's not about initialize the hardware SPI or I2C for sensors to work which can be initiated anytime when you want to use it. 

But for the Firebase library, it is about NTP time server data acquisition for time setting,  token preparation, generation and signing and Firebase authentication as same as in the app initialization in other platforms e.g. Android and Web which these process should done first before anything and assume that the internet is available during that processes because it requires for data exchanges between client and server.

You should design your program to meet the Firebase requirement.

Niltonsf, 2021-02-10:
@mobizt 
I'm understanding I think.

1. So I should call the initialization (begin) before everything (setup), this means that I won't be able to use it inside any function.

2. I won't be able to set the email and password from the data I receive from bluetooth, I have to manually insert the email and password inside the code?

3. I can't have a function like this then:
```
void funcClass::create_document_database(char email[], char password[]){
  int count = 0;

  delay(1000);
  
  SerialBT.disconnect();

  Serial.println("Connecting to WiFi");
  WiFi.mode(WIFI_STA);
  WiFi.begin("dd", "ssd");

  unsigned long startAttemptTime = millis();

  while (WiFi.status() != WL_CONNECTED && millis() - startAttemptTime < WIFI_TIMEOUT_MS)
  {
    Serial.print(".");
    delay(500);
  }

  if (WiFi.status() != WL_CONNECTED){
    Serial.println("\nConnection FAILED!");
    delete_file_info();
  }else{
    Serial.println("\nSuccessful Connection!");
  }

  auth.user.email = "tst@test.com.br";
  auth.user.password = "tst";

  config.host = FIREBASE_HOST;
  config.api_key = API_KEY;

  Firebase.begin(&config, &auth);
  Firebase.reconnectWiFi(true);

  String content;
  FirebaseJson js;
  
  //We will create the nested document in the parent path "a0/b0/c0
  //a0 is the collection id, b0 is the document id in collection a0 and c0 is the collection id id in the document b0.
  //and d? is the document id in the document collection id c0 which we will create.
  String documentPath = "a0/b0/c0/d" + String(count);

  js.set("fields/count/integerValue", String(count).c_str());
  js.set("fields/status/booleanValue", count % 2 == 0);
  js.toString(content);
  
  count++;

  Serial.println("------------------------------------");
  Serial.println("Create a document...");

  if (Firebase.Firestore.createDocument(&fbdo, FIREBASE_PROJECT_ID, "" /* databaseId can be (default) or empty */, documentPath.c_str(), content.c_str()))
  {
      Serial.println("PASSED");
      Serial.println("------------------------------------");
      Serial.println(fbdo.payload());
      Serial.println("------------------------------------");
      Serial.println();
  }
  else
  {
      Serial.println("FAILED");
      Serial.println("REASON: " + fbdo.errorReason());
      Serial.println("------------------------------------");
      Serial.println();
  }
}
```

mobizt, 2021-02-10:
The library initialization can be placed in any function, and can call many times when required, **unless it should run once in the setup** to let the library know where are your config and auth data. 

In the setup, no matter the WiFi is ready or not, you should call begin once with config and auth.

After begin was called, the authentication may fail due to no internet connection or missing required config and auth data, don't worry, you can call begin again and again as you want to change the authentication method.

The required config and auth data for different authentication method, see the Authentication examples.  

In the case of Email/Password sign-in and custom and OAuth2.0 tokens, you should  set the token expires data to zero before call the begin (re-initialize) in your functions. This will force the current token to expired.

Example, in your function:

```cpp
config.signer.tokens.expires = 0;
Firebase.begin(&config, &auth);
```

Which you can call it when ever you want.

As the top statement, do not forget to call Firebase.begin once in the setup.

Niltonsf, 2021-02-10:
@mobizt 
I have attempted to do the following, I have ran the Firebase.begin() in the setup and then tried to run it again inside a function but it gives me the same error.

This is what I did:
```
#define FIREBASE_HOST_main "automa.firebaseio.com"
#define API_KEY_main "sdakndaskjdnaskjn218213"

FirebaseAuth auth;
FirebaseConfig config;

void setup() {
  Serial.begin(115200);

  Serial.println("Starting...");
  delay(2000);
  Serial.println("Successfully started");

  config.host = FIREBASE_HOST_main;
  config.api_key = API_KEY_main;

  auth.user.email = "noemail";
  auth.user.password = "nopass";

  Firebase.begin(&config, &auth);
  Firebase.reconnectWiFi(true);

  if (!SPIFFS.begin(true)){
   Serial.println("Error mounting SPIFFS");
   return;
  }else{
    Serial.println("SPIFFS successfully mounted");
  }

  funcclass.delete_file_info();

  funcclass.programStart();
}
```
for the function later on, that will run inside the loop:
```
void funcClass::create_document_database(char email[], char password[]){
  int count = 0;

  delay(1000);
  
  SerialBT.disconnect();

  Serial.println("Connecting to WiFi");
  WiFi.mode(WIFI_STA);
  WiFi.begin("Internet", "password");

  unsigned long startAttemptTime = millis();

  while (WiFi.status() != WL_CONNECTED && millis() - startAttemptTime < WIFI_TIMEOUT_MS)
  {
    Serial.print(".");
    delay(500);
  }

  if (WiFi.status() != WL_CONNECTED){
    Serial.println("\nConnection FAILED!");
    delete_file_info();
    return;
  }else{
    Serial.println("\nSuccessful Connection!");
  }

  config.host = FIREBASE_HOST;
  config.api_key = API_KEY;

  auth.user.email = "test@test.com.br";
  auth.user.password = "test";

  config.signer.tokens.expires = 0;

  Firebase.begin(&config, &auth);
  Firebase.reconnectWiFi(true);

  String content;
  FirebaseJson js;
  
  String documentPath = "a0/b0/c0/d" + String(count);

  js.set("fields/count/integerValue", String(count).c_str());
  js.set("fields/status/booleanValue", count % 2 == 0);
  js.toString(content);
  
  count++;

  Serial.println("------------------------------------");
  Serial.println("Create a document...");

  if (Firebase.Firestore.createDocument(&fbdo, FIREBASE_PROJECT_ID, "" /* databaseId can be (default) or empty */, documentPath.c_str(), content.c_str()))
  {
      Serial.println("PASSED");
      Serial.println("------------------------------------");
      Serial.println(fbdo.payload());
      Serial.println("------------------------------------");
      Serial.println();
  }
  else
  {
      Serial.println("FAILED");
      Serial.println("REASON: " + fbdo.errorReason());
      Serial.println("------------------------------------");
      Serial.println();
  }
}
```

mobizt, 2021-02-10:
I don't know about your class.

Please see this example code and see how to call and access the external vars within the class.

Please change the credentials in the code.

https://gist.github.com/mobizt/35f062da12cfc86f16da0cc8a2ed75a7

Niltonsf, 2021-02-10:
@mobizt 
The main problem is working with Bluetooth together, I noticed that when removing bluetooth it works fine. I was attempting to use SerialBT.end() & SerialBT.disconnect() before starting the connection with WiFi. But when attempting to use .end() I get the error that I posted above. Do you know any soulutions for that?

Niltonsf, 2021-02-10:
Just a little question,
I'm getting the data from bluetooth like this:
```
char email_char[email.length()] = "", password_char[password.length()] = "";
  
  email.toCharArray(email_char, email.length());
  password.toCharArray(password_char, password.length());
```
But when I call the function and pass email_char, I get invalid email

mobizt, 2021-02-10:
I think you should know ESP32 BLE and WiFi use the same radio module and it can't work both at the same time.

I post this many times when people open the issue about this.