mobizt/Firebase-ESP-Client #20: Create Document - Firestore failed

PrateekGoyal18, 2021-02-13:
Hey Mobizt, 

Just love your repos man!

I am trying to get ahold of the create document example, but getting this error:
------------------------------------
Create a document...
FAILED
REASON:
Can you please check this.
I am using Nodemcu v0.9

mobizt, 2021-02-13:
Invalid authentication info.

For Firestore examples, you need to add the API key to the example when you log in with Email/Password.
See the RTDB example for authentication reference.

mobizt, 2021-02-13:
Add this two lines
https://github.com/mobizt/Firebase-ESP-Client/blob/main/examples/RTDB/Basic/Basic.ino#L26
https://github.com/mobizt/Firebase-ESP-Client/blob/main/examples/RTDB/Basic/Basic.ino#L61

closed 2021-02-13 by mobizt

PrateekGoyal18, 2021-02-13:
I am now getting the error: 
```
FAILED
REASON: Missing or insufficient permissions.
```

My security rules are open as you gave in documentation:
service cloud.firestore {
  match /databases/{database}/documents {
    match /todos/{document=**} {
      allow read, write: if true;
    }
  }
}

mobizt, 2021-02-13:
Please read [this](https://github.com/mobizt/Firebase-ESP-Client#authentication).

PrateekGoyal18, 2021-02-13:
So I should get the Service accounts key JSON file?

mobizt, 2021-02-13:
You should enable the Email sign-in provider if you Email/Password sign-in method.

After enable that, you can create user from the console or from library.

Please read more about Firebase Authentication from Google Firebase document.

PrateekGoyal18, 2021-02-13:
Done all that. After that only getting the error.

mobizt, 2021-02-13:
Read Firebase doc