mobizt/Firebase-ESP-Client #22: Firestore update

giorgio80, 2021-02-17:
hello,

first many compliments for the library. 

As I'm using Firestore I would like to ask if stream , that in Firestore are realtime updates, are supported 

thanks

mobizt, 2021-02-17:
That feature is not available in the REST APIs this library used. 

Only RPC protocol (gRPC) supports this feature and the RPC in the embedded device is not supported as it based on http2 or partly support in ESP32 from Esspressif ([voice assistance sdk](https://github.com/espressif/esp-va-sdk)) but they don't provide the sources.

But you can use the Cloud Functions to do this on the backend e.g. the update event from document or collection changes can be used to set the RTDB data and you can stream that data in RTDB instead.

This approach is easy and suitable for small memory device which you can do may things in the Cloud Functions before giving the result.

The big update of this library which will available today provide the easy way to create and deploy the Cloud Functions and  supports the Google Cloud Storage which allows the large file upload and many bugs fixed.

closed 2021-02-17 by mobizt

giorgio80, 2021-02-17:
hello @mobizt 
thanks for the suggestion.
On other side , on classic real-time database, the stream object ( acting as realtime update) is basically polling continuously to the system ?

mobizt, 2021-02-17:
No polling in stream connection. 

The connection between client and server keeps open and server will push the data to client based on event without client request.

giorgio80, 2021-02-17:
Ou,  

just from a technical point  of view which is then difference with Firestore ? I mean there's no stream possibility with Firestore, am I wrong ?

mobizt, 2021-02-17:
You should read in the Firebase Doc.

Firestore REST APIs used by the library is missing the realtime monitoring capabilities. This available on other clients that Firebase SDK was install e.g. Web, iOS, Android, Java.

The embedded device can work with Firebase only via REST APIs.
https://github.com/mobizt/Firebase-ESP8266/issues/126
https://github.com/mobizt/Firebase-ESP8266/issues/76
https://stackoverflow.com/questions/64832636/does-firebase-firestore-rest-api-support-streaming

mobizt, 2021-02-17:
I don't know your experiences about Firestore and which client you ever used.
I recommend you to read the Firebase's Firestore doc about the concept, abilities and limitations.