mobizt/Firebase-ESP-Client #23: Looking for an example on usage of ArrayValue for Firebase.Firestore.patchDocument

rek, 2021-02-18:
I am trying to update a key with an array, but having some trouble. 

Can you please provide some example usage for this data type: https://firebase.google.com/docs/firestore/reference/rest/v1beta1/ArrayValue ?

My use case is adding logging data over time in batches of numbers.

mobizt, 2021-02-18:
If you mean you want to add the array member to the existing array, you can't do with the REST APIs.

If the field assigned as  updateMask existed in the document, it will overwrite.  You can't merge new field or nested field into some field unless overwrite it.

You may create the Cloud Function to do this.

closed 2021-02-18 by mobizt

rek, 2021-02-19:
Not to update array, just to add a new one.

This is the solution I found the in end:

```
  FirebaseJson entry1;
  entry1.add("integerValue", "1");
  FirebaseJson entry2;
  entry2.add("integerValue", "2");

  FirebaseJsonArray container;
  container.add(entry1);
  container.add(entry2);

  FirebaseJson log;
  log.set("fields/logs/arrayValue/values", container);

  String content;
  log.toString(content);
  Firebase.Firestore.createDocument(&fbdo, FIREBASE_PROJECT_ID, "", documentPath, content.c_str())
```

Produces the required:
```
{
  "name": "projects/xxxxx/logs/9V78U4kG6HgPYcnxUwAa",
  "fields": {
    "logs": {
      "arrayValue": {
        "values": [
          {
            "integerValue": "1"
          },
          {
            "integerValue": "2"
          }
        ]
      }
    }
  },
  "createTime": "2021-02-19T19:26:35.937576Z",
  "updateTime": "2021-02-19T19:26:35.937576Z"
}
```

Thanks anyway, this lib is great!