mobizt/Firebase-ESP-Client #24: Infinite loop using login with email and password

talesw13, 2021-02-22:
Good afternoon.
I'm having a problem connecting using the email and password method.
I'm stuck in this loop:
https://github.com/mobizt/Firebase-ESP-Client/blob/704a2959a649ffe2635ae05615d66c3372af9fb1/src/signer/Signer.cpp#L613

chunckBufiSise is always = zero, so it doesn't leave the loop.
I'm using the code ---> examples ---> Firebase_esp_client ---> authentications ---> Email_Password

will it be a bug, or am i doing something wrong?
  thanks.

mobizt, 2021-02-22:
It's not impossible to run infinite because it will timed out after 5 seconds.

You may provide the wrong Firebase credentials or your internet connection.

closed 2021-02-22 by mobizt

mobizt, 2021-02-23:
Update:
It can be happen but not the cases valid Firebase credentials which everything must be ok with that.
Check your Firebase AUTH and other credentials.