mobizt/Firebase-ESP-Client #32: HELP - REASON: INVALID_PASSWORD

giselybelloadur, 2021-03-11:
I'm trying to use the Stream example from the Firebase_ESP_Client library and the following error is happening:
  -> Set string ...
  -> FAILED
  -> REASON: INVALID_PASSWORD

Where do I need to validate the email and password to use in the code?

mobizt, 2021-03-11:
The user that you created in the project.

The users tab in Authentication, see the picture below
https://github.com/mobizt/Firebase-ESP-Client/blob/main/media/images/Enable_Email_Password_Provider.png

mobizt, 2021-03-11:
You can use the [Discussions](https://github.com/mobizt/Firebase-ESP-Client/discussions) for the question and help.

closed 2021-03-11 by mobizt

giselybelloadur, 2021-03-11:
Hello, thank you so much for replying.

I already did that, and put my email and email password in the project, but
the password is invalid. Why?

mobizt, 2021-03-11:
No thing else more than your trying to log in with user in other project instead of your provided project in the code.

Make sure you log in with user of that project and provide the valid host name (FIREBASE_AUTH) , in case you create  many projects. 

If your project host name (Fire

mobizt, 2021-03-11:
Please do not reply from email, please go to Github in case I edit the message.