mobizt/Firebase-ESP-Client #35: RTDB Stream (callback mode) missing updates

jorgelamb, 2021-03-19:
**Describe the bug**
We are using RTDB with callback mode for a project. When updates are sent very frequently (~100ms between updates) some of them are lost and never provided to the callback function.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to examples/RTDB/Stream_Callback.ino
2. Configure all settings
3. Run it. We are using ESP32 from az-delivery
4. From some other device connected to the same realtime database, send a few updates under /Test/Stream node. See code below used in a test page, which sends updates to /Test/Stream/a0 to /Test/Stream/a9 every 100ms:
```
for(let i=0; i<10; i++) {
  setTimeout(function() {
     firebase.database().ref(`/Test/Stream/a${i}`).set(firebase.database.ServerValue.increment(1));
  }, 100*i);
}
```
**Expected behavior**
The ESP32 callback should be called and output the 10 updates:
```
Stream Data1 available...
STREAM PATH: /Test/Stream
EVENT PATH: /a0
DATA TYPE: int
EVENT TYPE: put
VALUE: 1

[9 more updates skipped]
```

**Actual behavior**
Some updates are missing / skipped, and never get provided to the callback function. See full output for one test with 10 updates:
```
Stream Data1 available...
STREAM PATH: /Test/Stream
EVENT PATH: /a0
DATA TYPE: int
EVENT TYPE: put
VALUE: 1

Stream Data1 available...
STREAM PATH: /Test/Stream
EVENT PATH: /a4
DATA TYPE: int
EVENT TYPE: put
VALUE: 1

Stream Data1 available...
STREAM PATH: /Test/Stream
EVENT PATH: /a7
DATA TYPE: int
EVENT TYPE: put
VALUE: 1

Stream Data1 available...
STREAM PATH: /Test/Stream
EVENT PATH: /a9
DATA TYPE: int
EVENT TYPE: put
VALUE: 1
```

**Screenshots**
N/A, output provided above

**IDE and its version:**
arduino-cli alpha Version: 0.15.2 Commit: 0d4cb150 Date: 2021-02-09T13:02:49Z

**ESP32 Arduino Core**
 - esp32:esp32

**Additional context**
- This was tested with Stream_Callback (not removing updates from that code), but all callbacks should be sent between updates from Callback. Removing those doesn't solve the issue.
- Depending on the test different updates are missed.
- Changing timeout between updates to 500ms all updates are sent to the callback in my tests.
- Changing timeout between updates to 300ms some updates start to be lost, but less frequently than with 100ms.
- Other Firebase integrations (ie. web) get all the updates, so they are ok in Firebase.

mobizt, 2021-03-19:
This is not a bug or issue.

In the http stream, there are no guarantee and can't be guaranteed that all server data changes will push to client and client will get all changes.

Client is waiting for server's incoming data without sending request  All available payload data will be read from SSL buffer as the result you've got.  

This different than other protocol e.g. TCP socket and MQTT which data will not miss as it can resend.

In http stream, client doesn't know anything about incoming stream event data as it has only payload without header which means client doesn't know the length or size of payload which used to determine the validity of data.

mobizt, 2021-03-19:
Other clients e.g. web, Android and iOS use the Firebase SDK that connected to server trough the Google own protocol which can't compare to REST client which use http protocol.

closed 2021-03-19 by mobizt

jorgelamb, 2021-03-19:
Hi,

We have been testing Firebase Stream and in the tests all 10 updates from the other device are consistently retrieved:
```
curl -Lv -X GET -H "Accept: text/event-stream" https://[my-project].firebaseio.com/Test/Stream.json
```

That command always outputs all the values, even 
```
event: put
data: {"path":"/a0","data":22}

event: put
data: {"path":"/a1","data":22}

event: put
data: {"path":"/a2","data":22}

event: put
data: {"path":"/a3","data":22}

event: put
data: {"path":"/a4","data":22}

event: put
data: {"path":"/a5","data":22}

event: put
data: {"path":"/a6","data":22}

event: put
data: {"path":"/a7","data":22}

event: put
data: {"path":"/a8","data":22}

event: put
data: {"path":"/a9","data":22}
```

It might be that the data is delivered in multiple chunks or several updates in the same one, but it looks like the data will be there. If it's one chunk, it could generate multiple callback, while if it's multiple chunks it could keep track of the pending information to parse it when it's complete.

Is there a way to log all the payload processed by the library from Firebase Stream protocol? Maybe to Serial port, or connecting via a proxy?

mobizt, 2021-03-20:
I will fix this timing issue and update here.

mobizt, 2021-03-21:
The update of library v 2.0.9 is available with the issues fixed.

I tested with 100,000 times data storing iteration loop with no delay, no stream event data lost.