mobizt/Firebase-ESP-Client #36: BUG

Elblassy, 2021-03-27:
**Describe the bug**
streamAvailable() is returning false , first time it work fine but i install wifimanager library and doesn't work with it , then i delete wifimanager library and write the example only but not work again

**Expected behavior**
get true when change in realtime database

**IDE and its version:**
 - Arduino
 - Version 1.8.13

**ESP8266 Arduino Core SDK version**
 - Version 2.7.4

Elblassy, 2021-03-27:
When i downgrade to version 2.0.8 it work fine !

mobizt, 2021-03-27:
It's not related to the library version.

Set [this to false](https://github.com/mobizt/Firebase-ESP-Client/blob/ca40a630d14de5a2ad0b4471c8b10f177f39445e/examples/RTDB/Basic/Basic.ino#L68) when you want other library to control WiFi.

I'm not recommend to let other library to control the WiFi reconnection, this library has its proper way to reconnect the WiFi and the proper synchronization of operation.

When you use other library to control WiFi reconnection, it interfere this library to trying to reconnect WiFi and interrupt the stream.

I recommend use WiFi manager only for setup the WiFi credential once in the setup and disable it after that (don't let it polling in the loop).

mobizt, 2021-03-27:
It you have the issue or something that you want to ask, please use [Discussions](https://github.com/mobizt/Firebase-ESP-Client/discussions).

closed 2021-03-27 by mobizt

mobizt, 2021-03-27:
The library has non-blocking WiFi reconnection process.

It stops and re-connects the WiFi when WiFi is not ready every 10 seconds by using the WiFi function **reconnect** and this is the best solution that make the WiFi connection resuming 100% works in both ESP32 and ESP8266 instead of stop it and call the begin and wait for the connection state ready in the loop like other people do.

Elblassy, 2021-03-27:
thank you for your replay , but i already do what you said to me and doesn't work in last version of your library. i am not professional in arduino but that what happened to me.

mobizt, 2021-03-27:
It's not related to this library. 

If you get the conflicts when using the WiFiManager, contact the dev of that library or read its doc how to disable the WiFi status polling.

Elblassy, 2021-03-27:
no sir i delete all library and let your library only and doesn't work in last version

mobizt, 2021-03-27:
There are no changes in the connection process between version 2.0.8 and the latest version of this library.

mobizt, 2021-03-27:
The library works as normally as I test it often everyday.

Elblassy, 2021-03-27:
I don't know why it happened i will try to delete all thing and try from scratch , Thanks for your replays :D

mobizt, 2021-03-27:
You should check your router or WiFi AP instead if you have the internet connection problem.

mobizt, 2021-03-27:
If you use ESP32 avoid [Sparkfun ESP32 Thing board](https://github.com/mobizt/Firebase-ESP32#tested-devices).

mobizt, 2021-03-27:
If you use only this library and no WiFiManager and still get the internet connection problem in the stream resuming, check the WiFi AP as I said above.

The non standard x-tal of some board may cause the unstable WiFi connection and it usually disconnected.

In case of long USB cable between board and your PC, you also need to use large capacitor (>220 uF) connected between 3v3 and gnd on the board to prevent voltage drop during high inrush current occurred when WiFi is working (during data transmission).

Elblassy, 2021-03-27:
Now i face other problem prevent me to solve first problem i get "max token generation retry reached" , i change all firebase and restart my router and get same error :(

mobizt, 2021-03-27:
That's because the internet connection failed.

Please try to change the WiFi hotspot or using your cell phone or PC as WiFi hot spot or reset the router.

Elblassy, 2021-03-27:
You really save my day bro, now i will do what you said about first problem :)

mobizt, 2021-03-27:
I'm glad to know it works again.