mobizt/Firebase-ESP-Client #38: Delay required to avoid "token is not ready"

BladeBronson, 2021-04-01:
**Describe the bug**
I tried forever to make the [Beginner_Start_Here](https://github.com/mobizt/Firebase-ESP-Client/blob/main/examples/RTDB/Beginner_Start_Here/Beginner_Start_Here.ino) work, but kept getting "token is not ready".

I finally added a 3000ms delay here:
https://github.com/mobizt/Firebase-ESP-Client/blob/a602918881b50e0c7ebe60f36707245b9b6ae6bf/examples/RTDB/Beginner_Start_Here/Beginner_Start_Here.ino#L75

I'm not sure if there's a callback available for Firebase.begin. A 3000ms delay doesn't sound like the right way to fix this, but it does work around the problem. I'd love it if you could give this a look and see if there's a better way to address the problem. Thank you!

mobizt, 2021-04-01:
If you use delay to waiting long enough (usually not more that 5 seconds) and token is still not ready which means token generation may fail by some reason that you need to get the information from it by setting the callback function or read the token status info from provided function, **Firebase.authTokenInfo()**.

To prevent token generation issues you should enable the Email/Password Sign-in provider, provide the valid user Email/password of a user and Web API key that belong to your project that matches the project assigned in your sketch.

Please read [this](https://github.com/mobizt/Firebase-ESP-Client#authentication) for authentication and prerequisites.

To get the information of token generation status, there are 2 methods that you can use,  method 1, to set the callback function to get the token info or method 2, loop to check to get the token info.

Below code shows you the both 2 methods to get token status.

```cpp

#if defined(ESP32)
#include <WiFi.h>
#elif defined(ESP8266)
#include <ESP8266WiFi.h>
#endif
#include <Firebase_ESP_Client.h>

/* 1. Define the WiFi credentials */
#define WIFI_SSID "WIFI_AP"
#define WIFI_PASSWORD "WIFI_PASSWORD"

/* 2. Define the Firebase project host name (required) */
#define FIREBASE_HOST "PROJECT_ID.firebaseio.com"

/** 3. Define the API key
 * 
 * The API key (required) can be obtained since you created the project and set up 
 * the Authentication in Firebase console. Then you will get the API key from
 * Firebase project Web API key in Project settings, on General tab should show the
 * Web API Key.
 * 
 * You may need to enable the Identity provider at https://console.cloud.google.com/customer-identity/providers 
 * Select your project, click at ENABLE IDENTITY PLATFORM button.
 * The API key also available by click at the link APPLICATION SETUP DETAILS.
 * 
*/
#define API_KEY "API_KEY"

/* 4. Define the user Email and password that already registered or added in your project */
#define USER_EMAIL "USER_EMAIL"
#define USER_PASSWORD "USER_PASSWORD"

/* 7. Define the FirebaseAuth data for authentication data */
FirebaseAuth auth;

/* 8. Define the FirebaseConfig data for config data */
FirebaseConfig config;

/* The calback function to print the token generation status */
void tokenStatusCallback(TokenInfo info);

/* The helper function to modify the database rules (optional) */
void prepareDatabaseRules(const char *path, const char *var, const char *readVal, const char *writeVal);

/* The function to print the operating results */
void printResult(FirebaseData &data);

/* The helper function to get the token status string */
String getTokenStatus(struct token_info_t info);

/* The helper function to get the token type string */
String getTokenType(struct token_info_t info);

/* The helper function to get the token error string */
String getTokenError(struct token_info_t info);

void setup()
{
[... 120 lines cut ...]
        return "error";

    default:
        break;
    }
    return "uninitialized";
}

/* The helper function to get the token error string */
String getTokenError(struct token_info_t info)
{
    String s = "code: ";
    s += String(info.error.code);
    s += ", message: ";
    s += info.error.message.c_str();
    return s;
}

```

I'm apologize for this about the additional functions that may need for the user who is new to Firebase Authentication.

This library aim that user are ready to use the Google recommended method instead of deprecated or legacy method using database secret as in the original [FirebaseESP32](https://github.com/mobizt/Firebase-ESP32) and [FirebaseESP8266](https://github.com/mobizt/Firebase-ESP8266) libraries.

The above code is taken from the [Authentications example here](https://github.com/mobizt/Firebase-ESP-Client/blob/main/examples/Authentications/Email_Password/Email_Password.ino) which you can check others Auth methods in the [Authentications folder](https://github.com/mobizt/Firebase-ESP-Client/tree/main/examples/Authentications).

mobizt, 2021-04-01:
I have edited the above code, please copy and try it again.

mobizt, 2021-04-01:
The reason why the delay is not include in the examples sketches is the token generation is the long running process, usage time depends on the type of authentications.

For Auth2.0  and custom token tokens, the generation processes involved NTP time acquisition, signing and token exchanging, which the time usage depends on the server request/response time and fix time for signing.

The time usage in token generation is also stated in the the [README](https://github.com/mobizt/Firebase-ESP-Client#authentication) doc.

Then the using of delay to waiting the token to be ready is not guarantee the completion as it may fail due to invalid credentials or other reasons.

Manual read the token status from **Firebase.authTokenInfo()** or callback function is the proper way that user can do with token based authentication.

closed 2021-04-01 by mobizt

BladeBronson, 2021-04-02:
Excellent! Thank you so much. I just confirmed that these approaches work. I appreciate you for all the work you'd put into this.