mobizt/Firebase-ESP-Client #39: HELP Token is not ready

nathanRamaNoodles, 2021-04-02:
For some reason, custom tokens are taking a long time to be generated.  I checked my router, but it seems fine.  I tried using different esp32 boards, and they have the same issue.  On the first run, it's instantaneous; however, after reboot it takes around 2 minutes to be generated.  What can I do to make the process faster?

My general scheme for my project is as follows: 
1. I have an android app that creates a user and sends the UID to the esp32
2. esp32 uses the UID to generate a token.
3. esp32 waits for the "token ready" event in the callback
4. esp32 uses the real-time database.

Sometimes, another issue that occurs is that my firebase project's website gets slower and my phone takes a while to sign in to firebase while the esp32 is generating a custom token.

nathanRamaNoodles, 2021-04-02:
I decided to run Wireshark with the esp32 connected via ethernet.  I am using my linux computer in bridge mode, so that I can intercept the data and see the issue.
It appears the esp32 is stuck with requesting NTP packets from a server even after getting the `Destination unreachable (Port unreachable)` message.  How can we make the esp32 to request the DNS for another ntp server so that we can avoid the inifinite loop.
[image]
In my packet trace, the esp32 eventually gets a connection from another ntp server:
[image]

mobizt, 2021-04-03:
The time usage is already stated in the [doc](https://github.com/mobizt/Firebase-ESP-Client#authentication).

You can set the device time yourself by acquiring time from NTP or set by RTC.

If time is ready before calling begin, the internal NTP time acquisition will be skipped.

closed 2021-04-03 by mobizt

nathanRamaNoodles, 2021-04-03:
Thank you, and I have good news!  I got it to work all the time now!
Maybe others would like to know too:
```
struct tm timeinfo;
while (!getLocalTime(&timeinfo)) {
  Serial.println("Setting time zone");
  configTime(-6 * 3600, 3600, "pool.ntp.org");
}
Firebase.begin(&config, &auth);
...
```

mobizt, 2021-04-03:
The valid time is required for internal JWT token signing and exchanging.

With user/password sign-in, the time is not use in the token generation process.