mobizt/Firebase-ESP-Client #41: BUG

bensalcie, 2021-04-06:
Here is an error I get after running the basic rtdb , any guide on this please?
Thanks.
In file included from C:\Users\Dev. Ben Salcie\Documents\Arduino\libraries\Firebase-ESP-Client-main\src/common.h:45:0,

                 from C:\Users\Dev. Ben Salcie\Documents\Arduino\libraries\Firebase-ESP-Client-main\src/Utils.h:37,

                 from C:\Users\Dev. Ben Salcie\Documents\Arduino\libraries\Firebase-ESP-Client-main\src/signer/Signer.h:37,

                 from C:\Users\Dev. Ben Salcie\Documents\Arduino\libraries\Firebase-ESP-Client-main\src/Firebase_ESP_Client.h:37,

                 from C:\Users\Dev. Ben Salcie\Documents\Arduino\libraries\Firebase-ESP-Client-main\examples\RTDB\Basic\Basic.ino:18:

C:\Users\Dev. Ben Salcie\Documents\Arduino\libraries\Firebase-ESP-Client-main\src/wcs/esp8266/FB_HTTPClient.h:50:30: fatal error: CertStoreBearSSL.h: No such file or directory

 #include <CertStoreBearSSL.h>

                              ^

mobizt, 2021-04-06:
You should update ESSP8266  Arduino Core SDK to the latest version.

closed 2021-04-06 by mobizt

bensalcie, 2021-04-07:
> Here is an error I get after running the basic rtdb , any guide on this please?
> Thanks.
> In file included from C:\Users\Dev. Ben Salcie\Documents\Arduino\libraries\Firebase-ESP-Client- [...]

> You should update ESSP8266 Arduino Core SDK to the latest version.

Thanks a lot Bro for the fast response and the time. Next time you could consider stating this in the documentation. Thanks a lot

mobizt, 2021-04-07:
The ESP8266 SDK you're using is almost 2-3 years released as it used axTLS as SSL engine instead of the later version that has bot axTLS and BearSSL.

mobizt, 2021-04-07:
This library is the unified and upgrade version of FirebaseESP8266 and FirebaseESP32 libraries which relied on BearSSL and mbedTLS libraries which required for token signing.

The doc of FirebaseESP8266 library also state the supported ESP8266 SDK version. Everyone that migrates from that library is already known.