mobizt/Firebase-ESP-Client #42: GCS - remote filename encoding for subfolders

WouterJansen, 2021-04-09:
When trying to upload to Firebase Google Storage and trying to use a remote filename with a subfolder goes wrong on the URL encoding part it seems. For example:

```c

Firebase.GCStorage.upload(&fbdo, "my_storage_bucket", "test.jpg", mem_storage_type_sd, gcs_upload_type_resumable, "subfolder/test.jpg", "image/jpeg", nullptr, nullptr, nullptr, gcsUploadCallback);`

void gcsUploadCallback(UploadStatusInfo info)
{
    if (info.status == fb_esp_gcs_upload_status_complete)
    {
        Serial.printf("BEFORE: %s\n", info.remoteFileName.c_str());
        Serial.printf("AFTER: %s\n", meta.name.c_str());
    }
    else if (info.status == fb_esp_gcs_upload_status_error)
    {
        Serial.printf("REASON: %s\n", info.errorMsg.c_str());
    }
}
```

This would give output like:
```
BEFORE: subfolder/test.jpg
AFTER: subfolder%2ftest.jpg
```

Am I doing something wrong or is this a bug?

mobizt, 2021-04-10:
I did not get error about this issue.

Please make sure you're using the latest version library.

closed 2021-04-10 by mobizt

WouterJansen, 2021-04-10:
Well it didn't work on new or older versions of the library. So I added
```c
|| c == '/'
```
to 
https://github.com/mobizt/Firebase-ESP-Client/blob/52e29812b23357d354fa51d866af62027c2126f3/src/Utils.h#L373

Just posting here if anyone else runs into this issue.

mobizt, 2021-04-10:
It's not the issue about url encode. I confirmed.

mobizt, 2021-04-10:
I already test and nothing to edit the function.

Please find the actual root cause which in your string construction and handling. 

The above code will return "File IO error" based on your file path is not valid in file system "test.jpg", it should be "/test.jpg".

WouterJansen, 2021-04-10:
Yeah the `test.jpg` should indeed be `/test.jpg`. The code above is just sample code to show the issue. I fixed it. 
I don't do any string construction, I just pass a string directly as argument to `Firebase.GCStorage.upload()` as in the sample above and as is done in the examples. 

Anyway, it now works for me after my fix. I don't care what you do with it. Just wanted to let anyone with similar issues know that this could be a **potential** fix for them.