mobizt/Firebase-ESP-Client #44: ENHANCEMENT

varunpvp, 2021-04-14:
Is it possible to implement onDisconnect for Firebase RTDB?

Hi @mobizt, thanks for such an amazing library.

I was wondering if it's possible to add onDisconnect functionality to Firebase RTDB.

If we're using websocket. It should be possible.

Let me know what you think.

Thanks.

mobizt, 2021-04-14:
It's not available because the library works with REST API or http request and response.

varunpvp, 2021-04-14:
I read the docs just now. I see that we use http stream header for getting real-time updates.

Anyway, let me know what's your take on this.

Thanks.

mobizt, 2021-04-14:
Stream is the http request which server will push the changes to client which the http session is reused and keep open.

The client can't send any request to the server during the stream event data listening only listen (polling read the available data in SSL client rx buffer).

Please search the old Firebase library for ESP8266 and ESP32 issues which I explain a lot if this kind of question or read the Firebase REST API doc.

closed 2021-04-14 by mobizt

mobizt, 2021-04-14:
For checking the disconnect status of  device (indirect method)

Let the device store the timestamp to the database at some fix interval (1-3 minutes).

Check that timestamp of device stored in the database by other apps or cloud functions.

varunpvp, 2021-04-14:
Yes, I know this once.

Thanks.