mobizt/Firebase-ESP-Client #46: How to use FCM messaging and real time database at same time?

AndrewKirkovski, 2021-04-17:
I'm trying to use both in project. Both work perfectly fine separately, but as soon as i use real time database and enable service account at same time, writing to rtdb starts to fail with "not found" in error reason 

disabling service account makes it work again, but messages are not being sent

I've also tried legacy FCM messages and that combo seem to work... except messages are never reaching the phone

mobizt, 2021-04-17:
With HTTPv1 FCM, you need to use OAuth2.0 sign-in method with service account.

Which you need to call Firebase.begin like this
https://github.com/mobizt/Firebase-ESP-Client/blob/52e29812b23357d354fa51d866af62027c2126f3/examples/Messaging/Send/HTTPv1/Send_Message/Send_Message.ino#L76-L83

With OAuth2.0 sign-in above, your device is now access or sign in as server admin, you can send HTTPv1 FCM message, access RTDB and all other Google and Firebase services provided by this library.

With legacy FCM you **need the server key from your project console only**. This key is not related to Firebase services. 
It initialized with
https://github.com/mobizt/Firebase-ESP-Client/blob/52e29812b23357d354fa51d866af62027c2126f3/examples/Messaging/Send/Legacy_API/Send_Message/Send_Message.ino#L63

It's not found to be issue when you use Firebase Data object in different class member functions e.g. Firebase.FCM, Firebase.RTDB, Firebase.GCS, Firebase.Firestore, Firebase.Storage and Firebase.GCStorage or Firebase.Functions.

All services have different API endpoints (different servers). 

Then If one Firebase Data object shared for used in many classes functions. Only one socket connection is available at a time and other new request at that time will be rejected or return false from the function.

No conflicts between integration usage between these different Firebase services in the same code sketch or project.

You need to read the comment in the every examples which give details and requirement to run the example.

closed 2021-04-17 by mobizt

mobizt, 2021-04-17:
You need to provide your minimum working code of your integration usage which clean and easy to investigate.

mobizt, 2021-04-17:
404 Not Found is because **the specified Realtime Database was not found**.

Make sure you provide the valid firebase host in **FIREBASE_HOST** defined macro.

AndrewKirkovski, 2021-04-17:
So I have setup like so
```
#define FIREBASE_HOST "*.firebaseio.com"
#define API_KEY "***"
#define FIREBASE_PROJECT_ID "*"
#define FIREBASE_CLIENT_EMAIL "*"
const char PRIVATE_KEY[] PROGMEM = "-----BEGIN PRIVATE ** "

 // .....

    config.host = FIREBASE_HOST;
    config.api_key = API_KEY;

    auth.user.email = "*";
    auth.user.password = "*";

    /* Assign the sevice account credentials and private key (required) */
   config.service_account.data.client_email = FIREBASE_CLIENT_EMAIL;
    config.service_account.data.project_id = FIREBASE_PROJECT_ID;
    config.service_account.data.private_key = PRIVATE_KEY; 

    Firebase.begin(&config, &auth);
    
    Firebase.reconnectWiFi(true);

#if defined(ESP8266)
    //Set the size of WiFi rx/tx buffers in the case where we want to work with large data.
    fbdo.setBSSLBufferSize(1024, 1024);
#endif

    //Set the size of HTTP response buffers in the case where we want to work with large data.
    fbdo.setResponseSize(1024);

    //tiny, small, medium, large and unlimited.
    //Size and its write timeout e.g. tiny (1s), small (10s), medium (30s) and large (60s).
    Firebase.RTDB.setwriteSizeLimit(&fbdo, "tiny");
````

And then use samples from lib to send messages and write to FCM database
RTDB write works every time if I comment out `config.service_account.data` setup
If i uncomment it, it stops writing with "not found" error. Obviously messaging stops working if i comment `config.service_account.data`

So both work OK in this setup but not at same time. Should I use 2 different Firebase objects?

I'll prepare minimum example to demo

mobizt, 2021-04-17:
The followingcredentials are required for Email/Password sign-in (id token)
```cpp
config.host = FIREBASE_HOST;
config.api_key = API_KEY;
auth.user.email = USER_EMAIL;
auth.user.password = USER_PASSWORD;
```

The following credentials are required for OAuth2.0 sign-in (access token).
```cpp
config.host = FIREBASE_HOST;
config.service_account.data.client_email = FIREBASE_CLIENT_EMAIL;
config.service_account.data.project_id = FIREBASE_PROJECT_ID;
config.service_account.data.private_key = PRIVATE_KEY;
```

The following credentials are required for custom token (id token with custom claims).
```cpp
config.host = FIREBASE_HOST;
config.api_key = API_KEY;
config.service_account.data.client_email = FIREBASE_CLIENT_EMAIL;
config.service_account.data.project_id = FIREBASE_PROJECT_ID;
config.service_account.data.private_key = PRIVATE_KEY;
```

The following credential required for legacy token access (database secret).
```cpp
config.host = FIREBASE_HOST;
config.signer.tokens.legacy_token = FIREBASE_AUTH;
```

***HTTPv1 FCM, Google Cloud Storage and Cloud Functions for Firebase support OAuth2.0 sign-in method only.***

***Firestore and Storage accept OAuth2.0, Email/Password (id token), and  custom token (id token with custom claims) sign-in meythods.***

***Firebase RTDB accepted all sign-in method above.***

To use separated or shared Firebase Data object can be available and depending on your usage scope (global or locally) and available free heap.

Only Firebase Data object for RTDB stream that required to use as global scope object and must shared with other uses.

irebase Data object used for RTDB normal read/store operation can be shared with other uses.

Each time irebase Data object served to connect to server it creates the SSL client instance (session) and it consumes large memory for SSL resouces. 

Making many http requests at the same moment from many Firebase Data object should be avoided due to out of memory and crash as wdt reset.

Legacy FCM required only Server Key which is similar to RTDB uses legacy token (database secret).

You should read the Firebase document for authentication supports.

mobizt, 2021-04-17:
The method that you sign-in, the token that generated is widely used for all Firebase services which the library acts as an app in other platform that need to be initialized once.

This except for legacy FCM that used its special server key and RTDB that used database secret.

To used all Firebase services together you need to use OAuth2.0 sign-in method.

The sign-in method selection is not available in this library because of some credentials are shared or commonly required.

To enable specific sign-in method you need to provide the required credentials for that method. 

You can call Firebase.begin at anytime you want to change the authentication method but should avoid at this latest version because of some small memory leaks may occurred each time you called begin and it will fix in the next update.

mobizt, 2021-04-17:
The library is not allow async or multi-socket connection for internal SSL client in Firebase Data object because of memory limited and huge of SSL recourse data usage per each socket connection.

AndrewKirkovski, 2021-04-17:
So assuming I have only OAuth2.0 Enabled
````
 /* Assign the project host (required) */
    config.host = FIREBASE_HOST;
    // config.api_key = API_KEY;

    // auth.user.email = "admin//";
    // auth.user.password = "admin//";

    /* Assign the sevice account credentials and private key (required) */
    config.service_account.data.client_email = FIREBASE_CLIENT_EMAIL;
    config.service_account.data.project_id = FIREBASE_PROJECT_ID;
    config.service_account.data.private_key = PRIVATE_KEY;
    // Firebase.FCM.setServerKey(FIREBASE_FCM_SERVER_KEY);

    Firebase.begin(&config, &auth);
    //required for legacy HTTP API
    
    Firebase.reconnectWiFi(true);

#if defined(ESP8266)
    //Set the size of WiFi rx/tx buffers in the case where we want to work with large data.
    fbdo.setBSSLBufferSize(1024, 1024);
#endif

    //Set the size of HTTP response buffers in the case where we want to work with large data.
    fbdo.setResponseSize(1024);

    //tiny, small, medium, large and unlimited.
    //Size and its write timeout e.g. tiny (1s), small (10s), medium (30s) and large (60s).
    Firebase.RTDB.setwriteSizeLimit(&fbdo, "tiny");
````

I should be able to make "set" operations to RTDB?

I've tried doing that, commented out sending push notications completely and wrote a value in DB

I've got 
```
FAILED
REASON: not found
------------------------------------
````

Assuming RTDB should support OAuth2.0 and assuming absolutely same write operation works when using `Email/Password sign-in`, what can this error indicate? 

I even have set up rules like this:

```
{
  "rules": {
    ".read": true,
    ".write": true
  }
````

Minimal example:

````
/**
 * Created by K. Suwatchai (Mobizt)
 * 
 * Email: k_suwatchai@hotmail.com
 * 
 * Github: https://github.com/mobizt
 * 
 * Copyright (c) 2021 mobizt
 *
*/

[... 156 lines cut ...]
    }
    else
    {
        Serial.println("FAILED");
        Serial.println("REASON: " + fbdo.errorReason());
        Serial.println("------------------------------------");
        Serial.println();
    }
    
}
````
This gives following output with or without messaging used

------------------------------------
Send Firebase Cloud Messaging...
PASSED
{
  "name": "projects/coffeemcu/messages/0:1618671832740051%05e6ca2605e6ca26"
}

------------------------------------

FAILED
REASON: not found
------------------------------------

AndrewKirkovski, 2021-04-17:
So far it looks like RTDB write (without sending messages as same time) works fine with legacy token (database secret) and email/password

Imediately when switching to Oauth2.0 it stops working

mobizt, 2021-04-17:
I test your code and it works normally.

No issue found from the test.

Your possible problem is FIREBASE_HOST, make sure you get it right.

[image]

Or try to update the library.

You should add this to bypass when token is not ready .

```cpp
loop()
{

 if (Firebase.authTokenInfo().status != token_status_ready)
    return;

  delay(1000);
  updateStatus("/turnedOn", String(turnedOnAt));
  delay(1000);
  sendMessage("Message","Message",10000);

}
```

In setup, the line before Firebase.begin you can add the token generation callback like this 
```cpp

void setup()
{
    
    Serial.begin(115200);

    WiFi.mode(WIFI_AP_STA);

    WiFiMulti.addAP("Rabbit", "8....y");
    
    Serial.print("Connecting to Wi-Fi");
    while (WiFiMulti.run() != WL_CONNECTED)
    {
        Serial.print(".");
        delay(300);
    }
    Serial.println();
    Serial.print("Connected with IP: ");
    Serial.println(WiFi.localIP());
    Serial.println();

    config.host = FIREBASE_HOST;

    config.service_account.data.client_email = FIREBASE_CLIENT_EMAIL;
    config.service_account.data.project_id = FIREBASE_PROJECT_ID;
    config.service_account.data.private_key = PRIVATE_KEY;
    
    config.token_status_callback = tokenStatusCallback;

    Firebase.begin(&config, &auth);
    
    Firebase.reconnectWiFi(true);

#if defined(ESP8266)
    fbdo.setBSSLBufferSize(1024, 1024);
#endif

    fbdo.setResponseSize(1024);

    Firebase.RTDB.setwriteSizeLimit(&fbdo, "tiny");

}
  

void tokenStatusCallback(TokenInfo info)
{
    if (info.status == token_status_error)
    {
        Serial.printf("Token info: type = %s, status = %s\n", getTokenType(info).c_str(), getTokenStatus(info).c_str());
        Serial.printf("Token error: %s\n", getTokenError(info).c_str());
    }
    else
    {
        Serial.printf("Token info: type = %s, status = %s\n", getTokenType(info).c_str(), getTokenStatus(info).c_str());
    }
}
```

mobizt, 2021-04-17:
The OAuth2.0 token generation process is time consuming. 

It do many things behind and use a lot of cpu processing (math calculation in encryption and 256 bit RSA signing) power and memory especially in ESP8266.

You need to wait until it completed. In ESP8266 you should not use delay to wait the token to be ready as it blocked the internal schedule tasks.

Please read this [doc](https://github.com/mobizt/Firebase-ESP-Client#authentication).

mobizt, 2021-04-17:
Don't compare to the legacy token that token is already in the code.

AndrewKirkovski, 2021-04-17:
But I believe if `FIREBASE_HOST` was incorrect neither legacy or email/password method would work? Are there any other cases where "not found" error can occur?

I added waiting for token and it didn't do any difference

`````
Connecting to Wi-Fi........
Connected with IP: 192.168.1.129

Token info: type = OAuth2.0 access token, status = on initializing
Token info: type = OAuth2.0 access token, status = on signing
Token info: type = OAuth2.0 access token, status = on request
Token info: type = OAuth2.0 access token, status = ready
FAILED
REASON: not found
------------------------------------
````

mobizt, 2021-04-17:
The only problem is invalid credential you provided.

Check your service account, it should assign to your Firebase project. You should get it from the Firebase setting console.

There are many service accounts for many users and projects in IAM console of Google Cloud Platforms.

You should use the right one.

mobizt, 2021-04-17:
Service Account you provide is not belong to the project owner user.

AndrewKirkovski, 2021-04-17:
I've tried intentionally adding wrong host to legacy method and got
````
FAILED
REASON: 404 Not Found
------------------------------------
````

With correct host it looks like so for legacy token:

````
Connected with IP: 192.168.1.129

PASSED
PATH: /turnedOn
TYPE: string
ETag: 3OcCYPn6BP6lbeJ80arb4O3Xhp0=
VALUE: ------------------------------------

PASSED
PATH: /turnedOn
TYPE: string
ETag: 3OcCYPn6BP6lbeJ80arb4O3Xhp0=
VALUE: ------------------------------------
````

> The only problem is invalid credential you provided.

Ok, but why would cloud messages work than with these credentials? They need different permissions?

Also why would same firebase_host work with different kind of token?

I even added owner role to FIREBASE_CLIENT_EMAIL in console, still no difference
[image]

mobizt, 2021-04-17:
Database secret is the privileges' key to bypass authentication. No authentication applied or checked by server.

Database security rules don't guard anything when you use database secret as it bypass all security.

That's why Goggle convince you to use new auth method by set it deprecated.

mobizt, 2021-04-17:
Again you should read [this](https://github.com/mobizt/Firebase-ESP-Client#authentication) I ever post.

At the top of authentication examples also provide the info how to get the credentials.

Please use your time to read the document.

AndrewKirkovski, 2021-04-17:
Still I can use legacy token or email/pass it to verify FIREBASE_HOST is set correctly. And it does write to same FIREBASE_HOST with these method. I only use it for sanity check, that copy pasting FIREBASE_HOST is correct. 

I believe oauth2.0 token would not be able to generate and send push messages successfully if it was set incorrectly? And yes, it sends messages to correct application. I used my time and read that document, all data is taken from there. How would cloud messaging work if it was wrong private key?
Or am I missing here something?

How it would be possible that both oauth 2.0 is working for messages correctly, host working with other methods, but not together? 

I can record a video showing where I take values from if words are not convincing enough

mobizt, 2021-04-17:
If you follow all the doc guides. I'm not sure the conflicts with your google user and project on the backend.

You can try with new Gmail sign up and use that email to create the fresh new Firebase account and try again.

It should work as I already test it with fresh new Gmail and Firebase project. No payment info added to the new project and everything works.

AndrewKirkovski, 2021-04-17:
Could I just email you video (I'll change project credentials later) so maybe you could see what's wrong?
There is no backend, I'm pretty much just writing in db based on basic example from repo

mobizt, 2021-04-17:
I mean google backend settings which google may block some service which used admin right access which we don't know.

For example, if your recovery or contact email was [listed as data breach](https://haveibeenpwned.com/).

mobizt, 2021-04-17:
Try with new Gmail and Firebase project first.  It should work.

AndrewKirkovski, 2021-04-17:
Unfortunately new project with other gmail account produces same error :/

mobizt, 2021-04-17:
Strange and  you may make something not correct.

mobizt, 2021-04-17:
Please try following (temporary add the access token into the library) 

Go to https://developers.google.com/oauthplayground/

Under Step 1Select & authorize APIs
Scroll to Firebase Cloud Messaging API v1 and select https://www.googleapis.com/auth/cloud-platform as step 1 in the image.

[image]

Scroll to Google OAuth2 API v2 and select https://www.googleapis.com/auth/userinfo.email and https://www.googleapis.com/auth/userinfo.profile
openid as in step 2 in image.

[image]

In the input box as in step 3 in image, copy and paste following text to it.

https://www.googleapis.com/auth/userinfo.email https://www.googleapis.com/auth/firebase.database

Click Authorize API buttion as in step 4 in image.

Follow the consent screen.

Click Exchange authorization code for token buttion as in step 5 in image.
[image]

Copy the value of access_token key as in step 6 in image.

[image]

Add the following code to line 1793 of Signer.cpp. (before line if (config->signer.preRefreshSeconds...)

    config->signer.tokens.access_token = "xxxxxxxxxxx";
    config->signer.tokens.token_type = token_type_oauth2_access_token;
    config->signer.tokens.expires = time(nullptr) + 3600;
    return;

https://github.com/mobizt/Firebase-ESP-Client/blob/52e29812b23357d354fa51d866af62027c2126f3/src/signer/Signer.cpp#L1793-L1794

Replace xxxxxxxxxxx with the access token.

Compile and upload.

Please undo the change of Signer.cpp after test.

It should work with this hack (token expired in 60 minutes)

mobizt, 2021-04-17:
Above test will prove that your email account has the access rights (owner permission) for the Firebase project.

If above test works, it means that the token generation of library is ok unless wrong credential provided.