mobizt/Firebase-ESP-Client #47: Multiple Errors

Abhesheksh, 2021-04-17:
Facing quite a lot of errors, pretty sure I am missing something major. Please help.

My code : 

`#include <common.h>
#include <FirebaseFS.h>
#include <Firebase_ESP_Client.h>
#include <Utils.h>

#include <ESP8266WebServer.h>
#include <WiFiManager.h>
#include <FirebaseESP8266.h>
//#include <ESP8266Ping.h>
#include <EEPROM.h>
//#include <Firebase.h>

#define Firebase_Host "host";
#define API_KEY "Apikey";

#define Firebase_Project_ID "XXXXX-8b462";

#define USER_EMAIL "123@1234.com";
#define USER_PASSWORD "123456";

//Define Firebase Data object
FirebaseData fbdo;

FirebaseAuth auth;
FirebaseConfig config;

WiFiManager wifiManager;
// Set web server port number to 80
WiFiServer server(80);

// Variable to store the HTTP request
String header;

// Auxiliar variables to store the current output state
String output5State = "off";
String output4State = "off";
int PIDState;
long product_UID;
String product_name;

// Assign output variables to GPIO pins
const int output5 = 5;
const int output4 = 4;

void setup() {
  Serial.begin(115200);
  
  

  // Initialize the output variables as outputs
  pinMode(output5, OUTPUT);
  pinMode(output4, OUTPUT);
  pinMode(0, INPUT_PULLUP);
  // Set outputs to LOW
  digitalWrite(output5, LOW);
  digitalWrite(output4, LOW);
  
  // Uncomment and run it once, if you want to erase all the stored information
   //wifiManager.resetSettings();
  

  WiFi.mode(WIFI_STA);

  //WiFi.softAPConfig(IPAddress(192,168,1,19), IPAddress(192,168,1,1), IPAddress(255,255,255,0));
  // set custom ip for portal
  wifiManager.setAPStaticIPConfig(IPAddress(192,168,1,19), IPAddress(192,168,1,1), IPAddress(255,255,255,0));
  wifiManager.setSTAStaticIPConfig(IPAddress(192,168,1,99), IPAddress(192,168,0,1), IPAddress(255,255,255,0));
  
  // fetches ssid and pass from eeprom and tries to connect
  // if it does not connect it starts an access point with the specified name
  // here  "AutoConnectAP"
  // and goes into a blocking loop awaiting configuration
  wifiManager.setConfigPortalTimeout(900);
  wifiManager.setConnectTimeout(900);
  wifiManager.setCountry("IN");
  wifiManager.setWiFiAPHidden(false);
  wifiManager.autoConnect("STAX_AP");

  
  if(wifiManager.autoConnect() == true)
  {
    Serial.println("TRUE BOOLEAN CONNECT");

  Serial.println("Connected.");
  Serial.print("IP address:\t");
  Serial.println(WiFi.localIP());   
  product_UID = random(1, 2147483647);
  Serial.println(product_UID);
  product_name = "WIFI Socket";

   // server.begin();

 /* Assign the project host and api key (required) */
    config.host = Firebase_Host;
    config.api_key = API_KEY;

    /* Assign the user sign in credentials */
    auth.user.email = USER_EMAIL;
    auth.user.password = USER_PASSWORD;

    Firebase.begin(&config, &auth);
    Firebase.reconnectWiFi(true);

  
  }
  else
  {
    Serial.println("FALSE - NO CONNECT");
    wifiManager.resetSettings();
    
  }
  // or use this for auto generated name ESP + ChipID
  //wifiManager.autoConnect();
  
  // if you get here you have connected to the WiFi

}

void loop(){

}`

Errors-

[Error Messages.txt](https://github.com/mobizt/Firebase-ESP-Client/files/6329777/Error.Messages.txt)

AndrewKirkovski, 2021-04-17:
You should use only one of 

```
#include <Firebase_ESP_Client.h>
#include <FirebaseESP8266.h>
```

I think

mobizt, 2021-04-17:
As AndrewKirkovski c said, based on your post in this repo, you should remove this line
#include <FirebaseESP8266.h>

closed 2021-04-17 by mobizt

Abhesheksh, 2021-04-18:
Thank you @AndrewKirkovski 

Removing that line cleared it up.