mobizt/Firebase-ESP-Client #49: HELP ! Messagin Legacy_API Send_Mesaggin (Flutter. firebase_messaging: ^9.1.0)

Lorenzo-Garcia, 2021-04-21:
Hello, I have problems receiving the title and body of the notification in my Flutter App, from my ESP8266, using the example of inherited delivery method, however the data I do receive.
I / flutter (19818): Handling a background message 0: 1618972418215250% 0353096bf9fd7ecd
I / flutter (19818): {timestamp: 1609815454, unit: celsius, temp: 28}
I / flutter (19818): FlutterFire Messaging: An error occurred in your background messaging handler:
I / flutter (19818): NoSuchMethodError: The getter 'title' was called on null.
. From the FireBase console or from Node js I have no problems.
I am Spanish speaking and again I have actually signed up to try to solve this problem, so I apologize if I have not understood any community rules. Thank you very much community

mobizt, 2021-04-21:
You should update the library to latest version and try again.

mobizt, 2021-04-21:
You need to follow this example.
https://github.com/mobizt/Firebase-ESP-Client/blob/52e29812b23357d354fa51d866af62027c2126f3/examples/Messaging/Send/Legacy_API/Send_Message/Send_Message.ino#L103-L106

Lorenzo-Garcia, 2021-04-23:
Hello, I'm sorry to tell you that the problem has not been resolved. When analyzing the message with breakpoint I find the following: The libraries on the ESP8266 are in their latest version, in the App too. In addition, it should be clarified that from the Node js script I can receive without problems.
category: null
contentAvailable: false
data: Map (3 items)
 > 0: "timestamp" -> "1609815454"
 > 1: "unit" -> "celsius"
 > 2: "temp" -> "28"
from: "611618541995"
messageId: "0: 1619149528235791% 0353096bf9fd7ecd"
messageType: null
mutableContent: false
notification: null
senderId: null
sentTime: DateTime (2021-04-23 00: 45: 28.232)
threadId: null

Lorenzo-Garcia, 2021-04-23:
I leave my code to be able to analyze it. Although I have only inserted the Token, Wifi credentials and removed comments and console impressions. Muchas gracias por contestar!
#if defined(ESP32)
#include <WiFi.h>
#elif defined(ESP8266)
#include <ESP8266WiFi.h>
#endif
#include <Firebase_ESP_Client.h>

#define WIFI_SSID "*"
#define WIFI_PASSWORD "*"

#define FIREBASE_FCM_SERVER_KEY "*"

#define DEVICE_REGISTRATION_ID_TOKEN "*"

FirebaseData fbdo;

unsigned long lastTime = 0;

int count = 0;

void sendMessage();

void setup()
{

    Serial.begin(115200);

    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
    Serial.print("Connecting to Wi-Fi");
    while (WiFi.status() != WL_CONNECTED)
    {
        Serial.print(".");
        delay(300);
    }

    //required for legacy HTTP API
    Firebase.FCM.setServerKey(FIREBASE_FCM_SERVER_KEY);

    Firebase.reconnectWiFi(true);

#if defined(ESP8266)
    //Set the size of WiFi rx/tx buffers in the case where we want to work with large data.
    fbdo.setBSSLBufferSize(1024, 1024);
#endif

    //Set the size of HTTP response buffers in the case where we want to work with large data.
    fbdo.setResponseSize(1024);

    sendMessage();
}

void loop()
{

    if (millis() - lastTime > 60 * 1000)
    {
        lastTime = millis();

        sendMessage();
    }
}

void sendMessage()
{

    FCM_Legacy_HTTP_Message msg;

    msg.targets.to = DEVICE_REGISTRATION_ID_TOKEN;

    msg.options.time_to_live = "1000";
    msg.options.priority = "high";

    msg.payloads.notification.title = "Notification title";
    msg.payloads.notification.body = "Notification body";
    msg.payloads.notification.icon = "myicon";
    msg.payloads.notification.click_action = "OPEN_ACTIVITY_1";

    FirebaseJson json;
    String payload;

    //all data key-values should be string
      json.add("temp", "28");
    json.add("unit", "celsius");
    json.add("timestamp", "1609815454");
    json.toString(payload);
    msg.payloads.data = payload.c_str();

    if (Firebase.FCM.send(&fbdo, &msg)) //send message to recipient
    {

        Serial.println("PASSED");
        Serial.println(Firebase.FCM.payload(&fbdo));
    }
    else
    {
        Serial.println("FAILED");
        Serial.println("REASON: " + fbdo.errorReason());
    }

    count++;
}

mobizt, 2021-04-23:
The library works as normally and test before published. 

I also test with Android and Web client again before I post this comment to confirm.

The result is already show the status of your message sending.
 **Serial.println("REASON: " + fbdo.errorReason());**

mobizt, 2021-04-23:
I'm apologize for this issue, this is my fault after update this issue in the Firebase-ESP32 and Firebase-ESP8266 Client libraries and forgot to update in this Firebase ESP Client library.

Please replace this message/FCM.cpp
https://github.com/mobizt/Firebase-ESP-Client/blob/main/src/message/FCM.cpp

with this gist version of FCM.cpp 
https://gist.github.com/mobizt/56a4868478d9d16e7682dbf2f4de9a8d

The next library update will fix with this updated file.

mobizt, 2021-04-30:
The [library update v2.1.0](https://github.com/mobizt/Firebase-ESP-Client/releases/tag/v2.1.0) is available with the bugs fixed.

closed 2021-04-30 by mobizt