mobizt/Firebase-ESP-Client #55: Firestore: Store Float/Double value | create document with auto id

fellnerse, 2021-05-06:
Hey, thanks for providing this tool, it's really awesome!

I just struggle with two points:
Running the Create_Documents.ino works fine for me. But now I want to store float/double values instead of [int/bool](https://github.com/mobizt/Firebase-ESP-Client/blob/0705dd7f92c77d990c2d8303731c601ea75b6bcf/examples/Firestore/Create_Documents/Create_Documents.ino#L99). I could not find any documentation on that in your repository and the google docs. Could you help me out there?

Also the 2nd question is regarding creating a new document. In your example you have a counter and provide firebase with your own unique id. In my case I want to have multiple devices writing to the same Firebase and need to use the auto id generation therefore. Could you tell me how the syntax is for this here as well? Could not find this in the docs either.

Do you have a general approach of transferring js code syntax to your plugin? I'm quite familiar using firestore with js...

Thanks in advance!

Keep the good work up!

mobizt, 2021-05-06:
For question, please use [Discussions](https://github.com/mobizt/Firebase-ESP-Client/discussions).

The library used REST APIs that provided by Google which lacks of many features as compare to other platforms which won't be used in embedded environment.

The supported field value types are stated [here](https://firebase.google.com/docs/firestore/reference/rest/v1/Value).

For the unique id, you can create random string or use UUID which is the way that RTDB crate the random parent node when push or post new data. 

On the Github you can search for [UUID](https://github.com/search?q=esp+uuid).

fellnerse, 2021-05-06:
Thanks for the quick response!

Cool that helps, maybe you want to add this link to your readme (I couldn't find it).

If I'm not mistaken you use this [endpoint](https://firebase.google.com/docs/firestore/reference/rest/v1/projects.databases.documents/createDocument). And there it says if the id is not provided the random id will be generated. Is this possible with your plugin, or would you still say I should go for my own ids?

mobizt, 2021-05-06:
For uuid, it depends on your requirement for choosing between your own manage uuid which you can add timestamp or anything in it and hash it or auto generated random uuid. 

The Firestore database structure is flexible and complex. I can't provide all documentation on this, the description and reference doc is available [here](https://github.com/mobizt/Firebase-ESP-Client/tree/main/src#firebase-cloud-firestore-functions). User need to know the concept of Firestore and familiar and used it fluently with his project.

Firestore is not suitable for small memory device because of large payload structure returned from a request. The risk on device crash due to out of memory is upon your user design the data structure. 

You also can't update the specific field as in other platforms, only replace with the new value is supported in REST APIs.

fellnerse, 2021-05-06:
For me random UUIDs are fine, so I will use the firebase provided ones.

You are completely right, but I just scanned your README on the frontpage of github; maybe it makes sense to highlight that you actually have a documentation of your code but in the source folder. The two small links are easily missed. Just an idea.
Also it looks like you autogenerate the README. Maybe you want to use Doxygen i.e.

In my case I do not need to do anything with the response, can I somehow just ignore the payload completely?

mobizt, 2021-05-06:
Thank for recommendation. The docs are written manually, and of course I will use that tool when I have time.

You can ignore the payload and clear all data and free the SSL client instance by calling clear from Firebase Data object.

`fbdo.clear();`

FYI: In Arduino platform, this kind of helper class was called library not a plugin.

closed 2021-05-06 by mobizt