mobizt/Firebase-ESP-Client #67: Issue with SD Card verification when performing Cloud Storage Upload

Fabrice-Beya, 2021-05-17:
**Describe the bug**
Whilst testing the firebase upload, i noticed i kept getting a File I/O error even though the SD card is working correctly in the rest of my application. When i comment out the sd_test condition from FCS.cpp the uploads works. 

**To Reproduce**
Steps to reproduce the behavior:
1. Configure application using email/password auth.
2. Copy and paste the upload code from the example
3. Attempt to upload a .txt file to project cloud storage bucket
4. Upload fails due to File I/O error.
5. Comment out the lines  FCS.cpp lines 232-236.
6. Upload is successful.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**IDE and its version:**
 - PlatformIO
 - Version 5.1.1

**ESP8266 Arduino Core SDK version**
 - Espressif 32 3.2.0

**Additional context**
When i comment out the check from FCS.cpp lines 232-236, and the upload worked.

mobizt, 2021-05-17:
Thanks for inform the bug.

Now it was fixed in the latest version, [v2.2.1](https://github.com/mobizt/Firebase-ESP-Client/releases/tag/v2.2.1).

Please update.

closed 2021-05-17 by mobizt

Fabrice-Beya, 2021-05-18:
Hi @mobizt 

Thanks for the quick response, confirm the issue is fixed.

Keep up the good work!

Regards