mobizt/Firebase-ESP-Client #68: eventType always returns put, also when database entry is deleted

casvanluijtelaar, 2021-05-18:
i'm trying to recognize in a rtd stream when an item gets added or deleted, but I can't tell which is which because eventtype is always "put"

[image]

casvanluijtelaar, 2021-05-18:
using the stream callback example

mobizt, 2021-05-18:
The event type was taken from Firebase server event payload. 

You should read this [Firebase RTDB API document](https://firebase.google.com/docs/reference/rest/database#section-streaming).

mobizt, 2021-05-18:
Only **put** and **patch** event types related to data changes event.

closed 2021-05-18 by mobizt

casvanluijtelaar, 2021-05-18:
so there is no way of knowing when a specific item is removed?

mobizt, 2021-05-18:
You should try and investigate, how it is going.

When you remove the node, the parsed response payload info are

```
stream path: /your/stream/node
event path: /path/to/deleted/node <----- where the deleted node relative to stream node path
event type: put <---  for set, push and delete functions while patch is for update node (updateNode) function.
data type: null <------ this states the data changes with delete
```

casvanluijtelaar, 2021-05-18:
that's what i'm looking for! dataPath stays the same but dataType is null. thanks!