mobizt/Firebase-ESP-Client #72: Firebase callFunction returns bad request when you add json data input.

Fabrice-Beya, 2021-05-22:
**Describe the bug**
When using the firebase call function, adding data input returns bad request response from firebase. Two variations of this are observed. 
1. Json data input returns bad request 400 result from firebase server.
2. When you just enter text and it succeeds, the req.body.data does not contain the data sent from the ESP32

Code:

FirebaseJson json;
    json.set("data/message", "Some text data example.");
    json.set("data/token", "Example of token data.");

    String data;
    json.toString(data, true);

    // Serial.println(data.c_str());

    if (Firebase.Functions.callFunction(&fbdo, FIREBASE_PROJECT_ID, PROJECT_LOCATION, "getAudio", data.c_str()))
    {
      Serial.printf("ok\n%s\n\n", fbdo.payload().c_str());
}

**To Reproduce**
Steps to reproduce the behavior:
1. Run code snipper example provoked above. You will get a bad request response.
2. Replace the data.c_str() with just a plain string constant, and you also get a bad request reply.

**Expected behavior**
Json objects converted into string constants should appear as part of req.body.data object in cloud functions.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**IDE and its version:**
 - PlatformIO - 5.1.1
 - Esspresif 32 3.2.0

**Additional context**
Can you also confirm if the library supports compatibility with both the cloud function onRequest and onCall functions.

mobizt, 2021-05-22:
You must  edit the json serialized string to keep the json tag to be parsed as string not serialized string.

For example
JSON object
**{\"a\":\"b\"}**

When pass to Firebase.Functions.callFunction
**"{"a":"b"}"** <-- invalid c char array or C++ string when used as payload 

Use escape sequences is not enough
**"{\\\"a\\\":\\\"b\\\"}"**   which will translate to string payload  **"{"a":"b"}"** <-- still not valid C/C++ string

It should be
**"{\\\\\\\"a\\\\\\\":\\\\\\\"b\\\\\\\"}"** which will translate as the string payload **"{\\\"a\\\":\\\"b\\\"}"** 

The functions argument will get string "{\\\"a\\\":\\\"b\\\"}"  which you need javascript JSON.parse to parse to JSON object like this.

**var obj = JSON.parse(stringPayload);**

Then use obj.message and obj.token in your cloud functions.

Then the solution is replace **"** in data payload with **\\\\\\\"** before pass to **Firebase.Functions.callFunction**

closed 2021-05-22 by mobizt

mobizt, 2021-05-22:
Above method may complicate I will add support to JSON type data in the next library update.

mobizt, 2021-05-22:
You can update the library to [v2.2.3](https://github.com/mobizt/Firebase-ESP-Client/releases/tag/v2.2.3) and check [this example](https://github.com/mobizt/Firebase-ESP-Client/blob/main/examples/CloudFunctions/CallFunction/CallFunction.ino) for how to.