mobizt/Firebase-ESP-Client #73: HELP - How could increment a value on firestore.

fooballs, 2021-05-22:
Hi, could be posible to perform on increment or decrement on an exist value on the firestore cloud.

i saw some examples like this, but is not found with this library.

firebase.firestore.FieldValue.increment(50)

firebase.firestore.FieldValue.increment(-50)

Regards.GC

mobizt, 2021-05-22:
No such function in REST APIs.

mobizt, 2021-05-22:
You should use Discussion for asking the question.

closed 2021-05-22 by mobizt

mobizt, 2021-05-22:
### Update:

You can use equivalent function, [commitDocument](https://github.com/mobizt/Firebase-ESP-Client/tree/main/src#commits-a-transaction-while-optionally-updating-documents) or [commitDocumentAsync](https://github.com/mobizt/Firebase-ESP-Client/tree/main/src#commits-a-transaction-while-optionally-updating-documents).

Please update the library to [v2.2.4](https://github.com/mobizt/Firebase-ESP-Client/releases/tag/v2.2.4) and check this new [IncrementFieldValue.ino](https://github.com/mobizt/Firebase-ESP-Client/blob/main/examples/Firestore/CommitDocument/IncrementFieldValue/IncrementFieldValue.ino) example for how to.