mobizt/Firebase-ESP-Client #84: Set the device time after Firebase.begin can cause the auth token expired error.

mobizt, 2021-06-14:
If you using Email/Password or OAuth2.0 authentication, the token will expired in 59 minutes as normal.

If the device time was set later by acquiring NTP or RTC, after **Firebase.begin** was called, the token expiry checking will not detect the changes then it may not refresh the token in-time and that expired token will use internally and the token expired error will show from Firebase calls.

This is the issue and will not be fixed as it can be prevent by setting the device time prior to **Firebase.begin** or set the config.time_zone to your local time zone prior to calling **Firebase.begin**.

**config.time_zone = xxxx;**

If you intend to change the device time later (after Firebase.begin), token needs to be refreshed once with this hacking config

```
config.signer.lastReqMillis = 0;
config.signer.tokens.expires = 0;
```

closed 2021-06-14 by mobizt

amitrao17, 2022-04-07:
I might be having a problem that relates to this...

The config.time_zone parameter is not discussed in the documentation unfortunately...

From a search of the source code, it appears to be an offset in hours. Why not make it an offset in seconds, as it is in the NTPClient.h library?

mobizt, 2022-04-07:
As you will see, this GMT offset type is float that gives the precision when convert to seconds when multiply by 3600 that is reason why I accept only GMT offset or timezone for time setting.

For example GMT + 4.30, you need to set this time_zone property to 4.5.

mobizt, 2022-04-07:
Only day light offset is set to 0 for internal time setting.

You can set the device system time manually prior to calling Firebase.begin, the library will not set the device system time when the time is valid or already set.

amitrao17, 2022-04-07:
Yes, I see that it is float. But the time_zone property could also be set to 4.5 * 3600 -- this gives the precision too.

mobizt, 2022-04-07:
I don't understand what you are trying to say.

You only need to set config.time_zone = 4.5. This value will pass to internal time setting functions and multiplied this value with 3600 to get the seconds offset at the following line.
https://github.com/mobizt/Firebase-ESP-Client/blob/22234540db87f643d7a173d7609dce94770dec12/src/FB_Utils.h#L1172

What is your problem?

mobizt, 2022-04-07:
@[amitrao17](https://github.com/amitrao17) If you have the issue, open the issue with the clean and minimal code and explain the issue description.

If you only have the alternative idea for time setting, I think its quite clear in the name and type of this current config property.