mobizt/Firebase-ESP-Client #91: RTDB callback dataType method returns unreadable data type

GChavezM, 2021-06-20:
**Describe the bug**
I am using the board nodeMCU (esp8266) and RTDB streams with callback, When I print the data type of the value, I am getting strange characters. The same behavior occurs also in the Callback example.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone or copy the examples/RTDB/DataChangesListener/Callback project
2. Configure the WIFI and Firebase credentials
3. Build and flash the project into a board
4. Open a serial monitor and see error

**Expected behavior**
In the serial monitor should appear the right data type of the payload (integer, float, double, boolean, string, JSON or blob)

**Actual behavior**
Instead, it prints incompressible characters
**Screenshots**
[image]

**IDE and its version:**
 -PlatformIO
 - Version 5.1.1

**ESP8266 Arduino Core SDK version**
 - Version 3.0.0

**Additional context**
My OS is Windows, using wsl1 (ubuntu 18.04)
Same results on macOS Big Sur Arduino IDE version 1.8.15

mobizt, 2021-06-20:
Thanks for report this and will fix it in incoming update.

mobizt, 2021-06-22:
The [latest library v2.3.0](https://github.com/mobizt/Firebase-ESP-Client/releases/tag/v2.3.0) is available with bugs fixed,  improves the memory usage and 8-10 k less memory consumption as compare to the previous version. 

In addition, classes can be excluded for disable or unused to save more memory via [FirebaseFS.h](https://github.com/mobizt/Firebase-ESP-Client/blob/v2.3.0/src/FirebaseFS.h#L40)

Please update the library.

closed 2021-06-22 by mobizt

CarlosEspinosaBrazil, 2021-06-24:
Help me please, in my case when I manually change the database it doesn't recognize string, returns empty, null. Thanks.