mobizt/Firebase-ESP-Client #92: HELP - with downloading large files from firebase storage using esp32

caravel-management, 2021-06-23:
I am trying to download large files (1-5 MB's) from my firebase storage onto an SD card through the esp32. When I use the examples for downloading files from firebase storage and cloud storage, only a very small fraction of the files actually download. However, when I try to upload the files from the SD card into firebase storage, everything works. 

I've read that the files will need to be sent over in chunks, but I do not know how to implement that or if this library is capable of doing that. Any help or pointers would be greatly appreciated!

mobizt, 2021-06-24:
Thanks for report this issue.
Now the issue was fixed in [v2.3.1](https://github.com/mobizt/Firebase-ESP-Client/releases/tag/v2.3.1), please update the library.

closed 2021-06-24 by mobizt

caravel-management, 2021-06-24:
@mobizt Oh awesome, it worked! Thank you! This library is amazing. Thank you for making it!