mobizt/Firebase-ESP-Client #96: HELP

DevyankShaw, 2021-07-02:
First of all, we want to say thank you @mobizt for bringing this library. It's making the development life easier. Currently, we are facing a problem in updating the _timestamp_  field of type _timestampValue_ in firestore document. 

We checked that _timestamp_  field of type _timestampValue_ is coming during Get document but we are not able to Update _timestamp_  field in the same type. Can you please suggest/share code snippet on how to update the _timestamp_ field using this library.

Below is the code snippet we are using:
```
void updateDataToFirestore() {

  timeClient.update();
  unsigned long unix_epoch = timeClient.getEpochTime(); //unix_epoch holds unix time in milliseconds

  String content;
  FirebaseJson js;

  String documentPath = "school inout/west bengal/howrah/711101/wb0001/2020-2021/students/inouts_id2";

  js.set("fields/timestamp/timestampValue", unix_epoch); //undefined reference issue coming during compile time
  js.set("fields/timestamp/timestampValue", DateTime(unix_epoch*1000));  // issue coming during compile time
  
  js.toString(content);

  Serial.println("------------------------------------");
  Serial.println("Update a document...");

  /** if updateMask contains the field name that exists in the remote document and 
   * this field name does not exist in the document (content), that field will be delete from remote document
   */

  if (Firebase.Firestore.patchDocument( & fbdo, FIREBASE_PROJECT_ID, "" /* databaseId can be (default) or empty */ , documentPath.c_str(), content.c_str(), "timestamp" /* updateMask */ )) {
    Serial.println("PASSED");
    Serial.println("------------------------------------");
    Serial.println(fbdo.payload());
    Serial.println("------------------------------------");
    Serial.println();
  } else {
    Serial.println("FAILED");
    Serial.println("REASON: " + fbdo.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }
}
```

mobizt, 2021-07-02:
The error is because you pass the unsupported value type, FirebaseJson supports int, float and double for number.

But that's not the point.

Timestamp value should be string not a number.
https://github.com/mobizt/Firebase-ESP-Client/blob/299bfd84e5296f76272cb34f1a3551bb9cd3fa90/examples/Firestore/CreateDocuments/CreateDocuments.ino#L117-L118

[Reference](https://firebase.google.com/docs/firestore/reference/rest/v1/Value)

DevyankShaw, 2021-07-02:
Okay got it!

It seems like you recently updated the examples. 

It's my bad I haven't checked it earlier. Anyway thanks, I will try the same.

closed 2021-07-02 by mobizt