mobizt/Firebase-ESP-Client #98: Not creating/updating a field in firestore document when path is long

DevyankShaw, 2021-07-04:
**Describe the bug**

I am trying to create/update _myTimestamp_ field in _inouts_id2_ document but it's showing as failed without any error message only when document path is long. Otherwise when document path is short, it's working fine.

**To Reproduce**

- String documentPath = "school inout/west bengal/howrah/711101/wb0001/2020-2021/students/inouts_id2";

    ```    
    Firebase Client v2.3.5
    
    Token info: type = id token, status = on request
    Token info: type = id token, status = ready
    Update a document... 
    FAILED
    REASON: 
    ------------------------------------
    ```

- String documentPath = "aa/bb";

    ```
    Firebase Client v2.3.5
    
    Token info: type = id token, status = on request
    Token info: type = id token, status = ready
    Update a document... 
    PASSED
    JSON DATA: 
    ------------------------------------
    
    ok
    {
      "name": "projects/electronics-f66ec/databases/(default)/documents/aa/bb",
      "fields": {
        "myTimestamp": {
          "timestampValue": "2021-07-04T08:56:23Z"
        }
      },
      "createTime": "2021-07-03T14:22:13.620444Z",
      "updateTime": "2021-07-04T09:09:07.522699Z"
    }
    ```

**Expected behavior**

String documentPath = "school inout/west bengal/howrah/711101/wb0001/2020-2021/students/inouts_id2";
```
Firebase Client v2.3.5

Token info: type = id token, status = on request
Token info: type = id token, status = ready
Update a document... 
PASSED
JSON DATA: 
------------------------------------

ok
{
  "name": "projects/electronics-f66ec/databases/(default)/documents/school inout/west bengal/howrah/711101/wb0001/2020-2021/students/inouts_id2",
  "fields": {
    "myTimestamp": {
      "timestampValue": "2021-07-04T08:56:23Z"
    }
  },
  "createTime": "2021-07-03T14:22:13.620444Z",
  "updateTime": "2021-07-04T09:09:07.522699Z"
}
```

**IDE and its version:**
 - Arduino
 - Version 1.8.15

**ESP8266 Arduino Core SDK version**
 - Version 3.0.1

**Additional context**

Below code snippet for the reference:
```
    String content;
    FirebaseJson js;

    String documentPath = "school inout/west bengal/howrah/711101/wb0001/2020-2021/students/inouts_id2";

    js.set("fields/myTimestamp/timestampValue", "2021-07-04T14:41:23+05:30");
    js.toString(content);

    Serial.println("Update a document... ");

    /** if updateMask contains the field name that exists in the remote document and
       this field name does not exist in the document (content), that field will be delete from remote document
    */

    if (Firebase.Firestore.patchDocument(&fbdo, FIREBASE_PROJECT_ID, "" /* databaseId can be (default) or empty */, documentPath.c_str(), content.c_str(), "myTimestamp" /* updateMask */))
    {
      Serial.println("PASSED");
      Serial.println("JSON DATA: ");
      Serial.println("------------------------------------");
      Serial.println();
      Serial.printf("ok\n%s\n\n", fbdo.payload().c_str());
    }
    else
    {
      Serial.println("FAILED");
      Serial.println("REASON: " + fbdo.errorReason());
      Serial.println("------------------------------------");
      Serial.println(); 
    }
```

mobizt, 2021-07-04:
It's not bug.

You are not show which device you use but I guess ESP8266.

You should increase the BearSSL buffer for rx and tx, like this

`fbdo.setBSSLBufferSize(2048, 1024); //2048 bytes for rx and 1024 bytes for tx.`

The default rx and tx buffers are  1024 and 512 by default and may be too small for large payload.

DevyankShaw, 2021-07-04:
Yes, the device is ESP8266.

Even after increasing BearSSL buffer size as suggested. Still, it's getting failed without error message.

mobizt, 2021-07-04:
You document path contains spaces, which should be avoid.

The space should be encoded to %20 string.

Library doesn't do extra works for path encoding for you and you should do it yourself.

closed 2021-07-04 by mobizt