mobizt/Firebase-ESP-Client #100: BUG RTDB.deleteNode gives error "data type mismatch" even though it does delete the node from firebase as expected

joshua-8, 2021-07-20:
**Describe the bug**
RTDB.deleteNode returns false and gives error "data type mismatch" even though it does delete the node from firebase as expected

**To Reproduce**
1. use `Firebase.RTDB.deleteNode` to delete a node from a Realtime Database
2. it returns false, and `fbdo.errorReason()` equals "data type mismatch" though the node I'm trying to delete does get deleted from the database

I will post an example demonstrating the issue below.

**Expected behavior**
I expected RTDB.deleteNode to return true when it functions properly and deletes from the database, not return false even when it functions. I would like to be able to detect when there is an issue with deleting from the database.

**IDE and its version:**
 - Arduino
 - Version 1.8.13

**ESP Arduino Core SDK version**
 - ESP32 1.0.6

code that demonstrates the issue:  (based on examples/RTDB/Basic)
```
#include <WiFi.h>
#include <Firebase_ESP_Client.h>

//Provide the token generation process info.
#include "addons/TokenHelper.h"
//Provide the RTDB payload printing info and other helper functions.
#include "addons/RTDBHelper.h"

/* 1. Define the WiFi credentials */
#define WIFI_SSID "removed"
#define WIFI_PASSWORD "removed"

/* 2. Define the API Key */
#define API_KEY "removed"

/* 3. Define the RTDB URL */
#define DATABASE_URL "something-default-rtdb.firebaseio.com" //<databaseName>.firebaseio.com or <databaseName>.<region>.firebasedatabase.app

/* 4. Define the user Email and password that alreadey registerd or added in your project */
#define USER_EMAIL "something@gmail.com"
#define USER_PASSWORD "removed"

//Define Firebase Data object
FirebaseData fbdo;

FirebaseAuth auth;
FirebaseConfig config;

unsigned long sendDataPrevMillis = 0;

int count = 0;

void setup()
{

  Serial.begin(115200);

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  Serial.printf("Firebase Client v%s\n\n", FIREBASE_CLIENT_VERSION);

  /* Assign the api key (required) */
  config.api_key = API_KEY;

  /* Assign the user sign in credentials */
  auth.user.email = USER_EMAIL;
  auth.user.password = USER_PASSWORD;

  /* Assign the RTDB URL (required) */
  config.database_url = DATABASE_URL;

  /* Assign the callback function for the long running token generation task */
  config.token_status_callback = tokenStatusCallback; //see addons/TokenHelper.h

  //Or use legacy authenticate method
  //config.database_url = DATABASE_URL;
  //config.signer.tokens.legacy_token = "<database secret>";

  Firebase.begin(&config, &auth);

  Firebase.reconnectWiFi(true);
}

void loop()
{
  if (Firebase.ready() && (millis() - sendDataPrevMillis > 15000 || sendDataPrevMillis == 0))
  {
    sendDataPrevMillis = millis();

    Serial.printf("Set int... %s\n", Firebase.RTDB.setInt(&fbdo, "/test/int", count) ? "ok" : fbdo.errorReason().c_str());
    delay(5000);
    Serial.printf("Delete node %s\n", Firebase.RTDB.deleteNode(&fbdo, "/test/int") ? "ok" : fbdo.errorReason().c_str());
    //the above line prints "Delete node data type mismatch" even though it deletes the value int.

    count++;
  }
}

```

result of running code:
```
Connecting to Wi-Fi........
Connected with IP: 10.0.0.14

Firebase Client v2.3.7

Token info: type = id token, status = on request
Token info: type = id token, status = ready
Set int... ok
Delete node data type mismatch
Set int... ok
Delete node data type mismatch

```
A value is added and deleted from the database, as I would expect.

Please ask me if there's any more information I could provide, and thank you so much for writing and supporting so many useful libraries!

mobizt, 2021-07-20:
Thanks, this issue will fix in the next update.

joshua-8, 2021-07-23:
I'm not trying to rush you. I'm just commenting to keep the issue open since stale bot closed it very quickly. Thank you for saying that this issue will be fixed in the next update!

mobizt, 2021-07-23:
Due to the major changes (bugs fixed, code refactoring, memory optimization), in the next update. 
I will inform you when the update is available.

joshua-8, 2021-07-23:
great! no rush! thank you!

closed 2021-07-26 by stale[bot]