mobizt/Firebase-ESP-Client #109: Bad request error appears randomly

Rakado93, 2021-07-29:
Hello, I am getting a bad request error when trying to connect to the stream data. Sometimes the code works and sometimes not (I recompiled the same code multiple times and sometimes I got stream data and sometimes I got a bad request error; no changes made neither to Firebase nor to code). To me it looks like after waiting some time and then recompiling the code it works again. I did not yet observe this problem when resetting the ESP32.

I am using the Arduino 1.8.13 IDE, an Espressif ESP32_DevKit_V4 and the Firebase ESP32 Client 3.10.5 library

`
  Firebase.begin("https://XYZ-default-rtdb.europe-west1.firebasedatabase.app/", "DatabaseSecret"); 
 
  Firebase.reconnectWiFi(true);

  if (!Firebase.beginMultiPathStream(firebaseData1, parentPath , childPath, childPathSize))
  {
    Serial.println("------------------------------------");
    Serial.println("Can't begin stream connection...");
    Serial.println("REASON: " + firebaseData1.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }

  Firebase.setMultiPathStreamCallback(firebaseData1, streamCallback, streamTimeoutCallback);

 
}
`
Thank you so much for any help to understand this problem, as it completely unexplainable to me!

mobizt, 2021-07-29:
You should use discussions instead.'
The error is not bug.

mobizt, 2021-07-29:
Please try the examples and follow it.

Read the doc and comment and recommendations in the examples.

closed 2021-07-29 by mobizt

mobizt, 2021-07-29:
You should update the ESP32 sdk and you  post to the wrong repository.

Rakado93, 2021-07-30:
Thank you and sorry for the confusion!