mobizt/Firebase-ESP-Client #121: I can't find any relevant program for streaming in firebase with esp32

pranavjasani, 2021-09-01:
Hello @mobizt please provide an example of firebase stream in esp32 to read data and control GPIO pins

mobizt, 2021-09-01:
Did you check [this](https://github.com/mobizt/Firebase-ESP-Client/tree/main/examples/RTDB/DataChangesListener)?

closed 2021-09-01 by mobizt

pranavjasani, 2021-09-01:
yes, but it gives API key invalid error with err code 400

mobizt, 2021-09-01:
Spend your time to read all the provided document carefully and try with all examples.

In all examples, there are detailed comments that guide and describe the purpose of each line of code which you should read it all.

For your code you should debug by yourself first.

If you try it enough and can't find the solution, you should provide the simple and clean code that you're already tried as well as the debug message that you've printed via serial monitor.

mobizt, 2021-09-01:
This is the [one of authentication example](https://github.com/mobizt/Firebase-ESP-Client/blob/main/examples/Authentications/SignInAsUser/EmailPassword/EmailPassword.ino) which gives you all details e.g. how you can take all required authentication credentials.

mobizt, 2021-09-01:
If you don't know about the Firebase Authentication which use the time to learn, you can try the legacy method for authenticate as [this example](https://github.com/mobizt/Firebase-ESP-Client/blob/main/examples/Authentications/LegacyTokenAuthen/LegacyTokenAuthen.ino).