mobizt/Firebase-ESP-Client #123: BUG - RTDB setTimestampAsync gives bad request while setTimestamp does not

gohjoseph, 2021-09-05:
**Describe the bug**
  `Firebase.RTDB.setTimestampAsync` gives error reason `bad request` while `Firebase.RTDB.setTimestamp` does not.

**To Reproduce**
```c++
bool ok = Firebase.RTDB.setTimestampAsync(&fbdo, "/test/timestamp");
if (!ok) Serial.printf("not ok timestamp: %s\n", fbdo.errorReason().c_str());
```
gives
```
not ok timestamp: bad request
```
as output in serial monitor

and database did not update while `setTimestamp` does update the database with no issue.

**Expected behavior**
Database should update with timestamp without error with `setTimestampAsync`

**IDE and its version:**
 - PlatformIO
 - Version 5.1.1 (Core)

**ESP8266 Arduino Core SDK version**
 - Version 3.2.0

**Additional context**
seems related to commit 856e5798fae7d48907addcc386c69b042143cceb

mobizt, 2021-09-06:
Thanks for report this bug.

Now it was fixed and please update the library to [v2.4.6](https://github.com/mobizt/Firebase-ESP-Client/releases/tag/v2.4.6).

gohjoseph, 2021-09-06:
Thank you for maintaining this library!

closed 2021-09-06 by gohjoseph