mobizt/Firebase-ESP-Client #139: FB_TCP_Client.h:85:99: error: '_use_insecure' was not declared in this scope

Razzorback225, 2021-10-02:
**Describe the bug**
When I'm trying to compile my code on the Arduino IDE, this gives me this error : 

`In file included from C:\Users\Thibaut\Documents\Arduino\libraries\Firebase_Arduino_Client_Library_for_ESP8266_and_ESP32\src/common.h:40:0,
                 from C:\Users\Thibaut\Documents\Arduino\libraries\Firebase_Arduino_Client_Library_for_ESP8266_and_ESP32\src/Utils.h:37,
                 from C:\Users\Thibaut\Documents\Arduino\libraries\Firebase_Arduino_Client_Library_for_ESP8266_and_ESP32\src/signer/Signer.h:37,
                 from C:\Users\Thibaut\Documents\Arduino\libraries\Firebase_Arduino_Client_Library_for_ESP8266_and_ESP32\src/Firebase.h:45,
                 from C:\Users\Thibaut\Documents\Arduino\libraries\Firebase_Arduino_Client_Library_for_ESP8266_and_ESP32\src/Firebase_ESP_Client.h:43,
                 from D:\Manul\ESP_Temp_Watchdog\CODE\ESP_Temp_Watchdog\ESP_Temp_Watchdog.ino:5:
C:\Users\Thibaut\Documents\Arduino\libraries\Firebase_Arduino_Client_Library_for_ESP8266_and_ESP32\src/wcs/esp32/FB_TCP_Client.h: In member function 'int FB_WCS::_connect(const char*, uint16_t, long unsigned int, long unsigned int)':
C:\Users\Thibaut\Documents\Arduino\libraries\Firebase_Arduino_Client_Library_for_ESP8266_and_ESP32\src/wcs/esp32/FB_TCP_Client.h:85:99: error: '_use_insecure' was not declared in this scope
     int ret = start_ssl_client(sslclient, host, port, _timeout, _CA_cert, NULL, NULL, NULL, NULL, _use_insecure);
`

I'm trying to compile for a custom board base on the ESP32 dev module

**Expected behavior**
Successful build

**IDE and its version:**
 - Arduino
 - Version 1.8.13

Razzorback225, 2021-10-02:
ESP32 core was not up to date

closed 2021-10-02 by Razzorback225

mobizt, 2021-10-03:
Thanks for report this.

This is actually issue which compile error in ESP32 Core v1.0.4 and earlier version.

The library updated and this bug fixed.

commit 4eec170f (2021-10-03): Fix #139 compilation error in ESP32 Core SDK v1.0.4 and earlier.