mobizt/Firebase-ESP-Client #169: start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!

escopecz, 2021-10-30:
**Describe the bug**
I cannot connect to Firebase using TTGO T-display with this minimal sketch:

platformio.ini:
```ini
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
monitor_speed = 115200
upload_speed = 921600
build_flags = 
	-Os
	-DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_DEBUG
	-DUSER_SETUP_LOADED=1
	-DST7789_DRIVER=1
	-DTFT_WIDTH=135
	-DTFT_HEIGHT=240
	-DCGRAM_OFFSET=1
	-DTFT_MISO=-1
	-DTFT_MOSI=19
	-DTFT_SCLK=18
	-DTFT_CS=5
	-DTFT_DC=16
	-DTFT_RST=23
	-DTFT_BL=4
	-DTFT_BACKLIGHT_ON=1
	-DLOAD_GLCD=1
	-DLOAD_FONT2=1
	-DLOAD_FONT4=1
	-DLOAD_FONT6=1
	-DLOAD_FONT7=1
	-DLOAD_FONT8=1
	-DLOAD_GFXFF=1
	-DSMOOTH_FONT=1
	-DSPI_FREQUENCY=40000000
	-DSPI_READ_FREQUENCY=6000000
	-D PIO_FRAMEWORK_ARDUINO_MMU_CACHE16_IRAM48_SECHEAP_SHARED
lib_deps = 
	bodmer/TFT_eSPI@^2.3.67
	adafruit/DHT sensor library@^1.4.3
	mobizt/Firebase Arduino Client Library for ESP8266 and ESP32@^2.5.5
```
main.cpp:
```cpp
#include <Arduino.h>
#include <TFT_eSPI.h> 
#include <SPI.h>
#include <WiFi.h>
#include <Firebase_ESP_Client.h>
#include <Wire.h>
#include "esp_adc_cal.h"
#include <Adafruit_Sensor.h>
#include <DHT.h>

//Provide the token generation process info.
#include <addons/TokenHelper.h>

const char *ssid = "[censored]";
const char *password = "[censored]";

#define API_KEY "[censored]"
#define FIREBASE_PROJECT_ID "[censored]"
#define USER_EMAIL "[censored]"
#define USER_PASSWORD "[censored]"

#define DHTTYPE DHT11

FirebaseData fbdo;
FirebaseAuth auth;
FirebaseConfig config;

DHT dht2(2, DHTTYPE);

unsigned long sendDataPrevMillis = 0;
unsigned long count = 0;

void setup()
{
  Serial.begin(115200);
  delay(1000);

  Serial.printf("Free heap: %d\n", ESP.getFreeHeap());

  Serial.println();
  Serial.println();
  Serial.printf("Connecting to: %s\n", ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(300);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  Serial.printf("Firebase Client v%s\n\n", FIREBASE_CLIENT_VERSION);

  /* Assign the api key (required) */
  config.api_key = API_KEY;

  /* Assign the user sign in credentials */
  auth.user.email = USER_EMAIL;
  auth.user.password = USER_PASSWORD;

  /* Assign the callback function for the long running token generation task */
  config.token_status_callback = tokenStatusCallback; //see addons/TokenHelper.h

  Serial.println("Firebase.begin start");
  Firebase.begin(&config, &auth);
  Serial.println("Firebase.begin end");
  Firebase.reconnectWiFi(true);

  dht2.begin();

  Serial.printf("Free heap: %d\n", ESP.getFreeHeap());
 
  delay(5000);
}

void loop()
{
  delay(2000);
  float h2 = dht2.readHumidity();
  float t2 = dht2.readTemperature();

  if (Firebase.ready() && (millis() - sendDataPrevMillis > 15000 || sendDataPrevMillis == 0))
  {
    String documentPath = "values";
    FirebaseJson content;
    content.set("fields/myDouble/doubleValue", 123.45678);
    Serial.println("Create a document... ");
    if (Firebase.Firestore.createDocument(&fbdo, FIREBASE_PROJECT_ID, "", documentPath.c_str(), content.raw())) {
      Serial.printf("ok\n%s\n\n", fbdo.payload().c_str());
    } else {
      Serial.println(fbdo.errorReason());
    }
  }
}
```

It builds like so:
```
> Executing task: platformio run <

Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.3.2) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10006.210326 (1.0.6) 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 32 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <TFT_eSPI> 2.3.70
|   |-- <SPIFFS> 1.0
|   |   |-- <FS> 1.0
|   |-- <FS> 1.0
|   |-- <SPI> 1.0
|-- <DHT sensor library> 1.4.3
|   |-- <Adafruit Unified Sensor> 1.1.4
|-- <Firebase Arduino Client Library for ESP8266 and ESP32> 2.5.5
|   |-- <WiFi> 1.0
|   |-- <FS> 1.0
|   |-- <SD(esp32)> 1.0.5
|   |   |-- <FS> 1.0
|   |   |-- <SPI> 1.0
|   |-- <SPIFFS> 1.0
|   |   |-- <FS> 1.0
|   |-- <WiFiClientSecure> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <SPI> 1.0
|   |-- <SD_MMC> 1.0
|   |   |-- <FS> 1.0
|-- <Adafruit Unified Sensor> 1.1.4
|-- <SPI> 1.0
|-- <WiFi> 1.0
|-- <Wire> 1.0.1
Building in release mode
Retrieving maximum program size .pio/build/esp32dev/firmware.elf
Checking size .pio/build/esp32dev/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  12.9% (used 42288 bytes from 327680 bytes)
Flash: [========  ]  79.6% (used 1043222 bytes from 1310720 bytes)
========================================================== [SUCCESS] Took 2.37 seconds ==========================================================

Terminal will be reused by tasks, press any key to close it.
```

And this is monitor output:
```
> Executing task: platformio device monitor <

--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at http://bit.ly/pio-monitor-filters
--- Miniterm on /dev/cu.usbserial-01DA6D7C  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Free heap: 327820

Connecting to: Orlik
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
.......[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:419] _eventCallback(): STA IP: 192.168.8.230, MASK: 255.255.255.0, GW: 192.168.8.1
.
WiFi connected
IP address: 
192.168.8.230
Firebase Client v2.5.5

Firebase.begin start
Token info: type = id token, status = on request
[I][ssl_client.cpp:127] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
Token info: type = id token, status = error
Token error: code: 400, message: bad request
Token info: type = id token, status = on request
[I][ssl_client.cpp:127] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
Token info: type = id token, status = error
Token error: code: 400, message: bad request
Token info: type = id token, status = on request
[I][ssl_client.cpp:127] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
Token info: type = id token, status = error
Token error: code: 400, message: bad request
Token info: type = id token, status = on request
[I][ssl_client.cpp:127] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
Token info: type = id token, status = error
Token error: code: 400, message: bad request
Token info: type = id token, status = on request
[I][ssl_client.cpp:127] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
Token info: type = id token, status = error
Token error: code: 400, message: bad request
Token info: type = id token, status = on request
[I][ssl_client.cpp:127] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
Token info: type = id token, status = error
Token error: code: 400, message: bad request
Token info: type = id token, status = error
Token error: code: -32, message: max token generation retry reached
Firebase.begin end
Free heap: 272960
Token info: type = id token, status = on request
[I][ssl_client.cpp:127] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
Token info: type = id token, status = error
Token error: code: 400, message: bad request
Token info: type = id token, status = on request
[I][ssl_client.cpp:127] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
Token info: type = id token, status = error
Token error: code: 400, message: bad request
Token info: type = id token, status = on request
[I][ssl_client.cpp:127] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
...
```
I've read in another issue that the warning can happen when heap memory is < 100k, but I have 272k. Is there anything else that a more experienced IoT colleague can spot from the info above?

**IDE and its version:**
 - Arduino, PlatformIO
 - Version Home 3.3.4

**ESP32 Arduino Core SDK version**
 - Version 5.2.2

mobizt, 2021-10-30:
This is not an issue.

The insecure! is warning from core WiFiClientSecure library which just warn you that no server verifications e.g. root certificate, private cert and key are not provided.

The error is because your project web api key may not valid.

mobizt, 2021-10-30:
The web api key, email and password is used for id token generation process (request from googleapi sevice). The id token used for later Firebase authentication.

closed 2021-10-30 by mobizt

escopecz, 2021-10-30:
That was it. Typo in the email address 🤦 

Thank you!