mobizt/Firebase-ESP-Client #172: timeout problems when with wifi and without internet

giselybelloadur, 2021-11-04:
I'm having problems when I'm without internet, for example, when I call the Firebase.RTDB.setString or Firebase.RTDB.getString routine the program is stuck for a long time, up to 2 minutes. This also happens if you call the Firebase.RTDB.readStream routine. How to interrupt the routine? How to identify that there is no internet before calling the routines? At no time I saw firebaseData.streamTimeout() error.

**IDE and its version:**
 - Arduino, PlatformIO
 - Version [e.g. 1.8.13]

** Esp32 mini

**Firebase_ESP_Client.h**
 -  2.3.7 and up

mobizt, 2021-11-04:
Please see the comments in the basic example that provides the options to set the various time out included this connection time out.

Please consider to use discussions for the question.

closed 2021-11-04 by mobizt

mobizt, 2021-11-04:
​  ​/*​* Timeout options, below is default config. 
  
 ​  //WiFi reconnect timeout (interval) in ms (10 sec - 5 min) when WiFi disconnected. 
 ​  config.timeout.wifiReconnect = 10 * 1000; 
  
 ​  //Socket begin connection timeout (ESP32) or data transfer timeout (ESP8266) in ms (1 sec - 1 min). 
 ​  config.timeout.socketConnection = 30 * 1000; 
  
 ​  //ESP32 SSL handshake in ms (1 sec - 2 min). This option doesn't allow in ESP8266 core library. 
 ​  config.timeout.sslHandshake = 2 * 60 * 1000; 
  
 ​  //Server response read timeout in ms (1 sec - 1 min). 
 ​  config.timeout.serverResponse = 10 * 1000; 
  
 ​  //RTDB Stream keep-alive timeout in ms (20 sec - 2 min) when no server's keep-alive event data received. 
 ​  config.timeout.rtdbKeepAlive = 45 * 1000; 
  
 ​  //RTDB Stream reconnect timeout (interval) in ms (1 sec - 1 min) when RTDB Stream closed and want to resume. 
 ​  config.timeout.rtdbStreamReconnect = 1 * 1000; 
  
 ​  //RTDB Stream error notification timeout (interval) in ms (3 sec - 30 sec). It determines how often the readStream 
 ​  //will return false (error) when it called repeatedly in loop. 
 ​  config.timeout.rtdbStreamError = 3 * 1000; 
  
 ​  ​*/