mobizt/Firebase-ESP-Client #199: Library not working in Platform IO

jaydip-pawar, 2021-11-30:
I am using this library in arduino IDE and it's work fine but when I start my project in Platform IO it's giving me error like this,

```
In file included from .pio\libdeps\esp32doit-devkit-v1\Firebase Arduino Client Library for ESP8266 and ESP32\src/Utils.h:37:0,
                 from .pio\libdeps\esp32doit-devkit-v1\Firebase Arduino Client Library for ESP8266 and ESP32\src\signer/Signer.h:37,
                 from .pio\libdeps\esp32doit-devkit-v1\Firebase Arduino Client Library for ESP8266 and ESP32\src\Firebase.h:45, 
                 from .pio\libdeps\esp32doit-devkit-v1\Firebase Arduino Client Library for ESP8266 and ESP32\src\Firebase.cpp:33:
.pio\libdeps\esp32doit-devkit-v1\Firebase Arduino Client Library for ESP8266 and ESP32\src/common.h:34:17: fatal error: SPI.h: No such file or directory

*************************************************************
* Looking for SPI.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SPI.h"
* Web  > https://platformio.org/lib/search?query=header:SPI.h
*
*************************************************************

compilation terminated.
*** [.pio\build\esp32doit-devkit-v1\lib329\Firebase Arduino Client Library for ESP8266 and ESP32\Firebase.cpp.o] Error 1
```
Then I tried adding SPI and other libraries got in error in my platformio.ini.

```
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
monitor_speed = 115200
lib_deps = 
    mobizt/Firebase Arduino Client Library for ESP8266 and ESP32 @ ^2.7.0
    SPI
    FS
    SPIFFS
    SD
```

After adding this I get following error

```
In file included from .pio\libdeps\esp32doit-devkit-v1\SD/utility/Sd2Card.h:26:0,
                 from .pio\libdeps\esp32doit-devkit-v1\SD/utility/SdFat.h:29,
                 from .pio\libdeps\esp32doit-devkit-v1\SD/SD.h:25,
                 from .pio\libdeps\esp32doit-devkit-v1\SD\File.cpp:15:
.pio\libdeps\esp32doit-devkit-v1\SD/utility/Sd2PinMap.h:371:2: error: #error Architecture or board not supported.
 #error Architecture or board not supported.
  ^
Compiling .pio\build\esp32doit-devkit-v1\lib28d\SD\utility\Sd2Card.cpp.o
In file included from .pio\libdeps\esp32doit-devkit-v1\SD/utility/SdFat.h:29:0,
                 from .pio\libdeps\esp32doit-devkit-v1\SD/SD.h:25,
                 from .pio\libdeps\esp32doit-devkit-v1\SD\File.cpp:15:
.pio\libdeps\esp32doit-devkit-v1\SD/utility/Sd2Card.h:65:37: error: 'SS_PIN' was not declared in this scope
 uint8_t const  SD_CHIP_SELECT_PIN = SS_PIN;
                                     ^
.pio\libdeps\esp32doit-devkit-v1\SD/utility/Sd2Card.h:68:31: error: 'MOSI_PIN' was not declared in this scope
 uint8_t const  SPI_MOSI_PIN = MOSI_PIN;
                               ^
.pio\libdeps\esp32doit-devkit-v1\SD/utility/Sd2Card.h:70:31: error: 'MISO_PIN' was not declared in this scope
 uint8_t const  SPI_MISO_PIN = MISO_PIN;
                               ^
.pio\libdeps\esp32doit-devkit-v1\SD/utility/Sd2Card.h:72:30: error: 'SCK_PIN' was not declared in this scope
 uint8_t const  SPI_SCK_PIN = SCK_PIN;
                              ^
In file included from .pio\libdeps\esp32doit-devkit-v1\SD/utility/Sd2Card.h:26:0,
                 from .pio\libdeps\esp32doit-devkit-v1\SD/utility/SdFat.h:29,
                 from .pio\libdeps\esp32doit-devkit-v1\SD\SD.h:25,
                 from .pio\libdeps\esp32doit-devkit-v1\SD\SD.cpp:53:
.pio\libdeps\esp32doit-devkit-v1\SD/utility/Sd2PinMap.h:371:2: error: #error Architecture or board not supported.
 #error Architecture or board not supported.
  ^
*** [.pio\build\esp32doit-devkit-v1\lib28d\SD\File.cpp.o] Error 1
In file included from .pio\libdeps\esp32doit-devkit-v1\SD/utility/SdFat.h:29:0,
                 from .pio\libdeps\esp32doit-devkit-v1\SD\SD.h:25,
                 from .pio\libdeps\esp32doit-devkit-v1\SD\SD.cpp:53:
.pio\libdeps\esp32doit-devkit-v1\SD/utility/Sd2Card.h:65:37: error: 'SS_PIN' was not declared in this scope
 uint8_t const  SD_CHIP_SELECT_PIN = SS_PIN;
                                     ^
.pio\libdeps\esp32doit-devkit-v1\SD/utility/Sd2Card.h:68:31: error: 'MOSI_PIN' was not declared in this scope
 uint8_t const  SPI_MOSI_PIN = MOSI_PIN;
                               ^
.pio\libdeps\esp32doit-devkit-v1\SD/utility/Sd2Card.h:70:31: error: 'MISO_PIN' was not declared in this scope
 uint8_t const  SPI_MISO_PIN = MISO_PIN;
                               ^
.pio\libdeps\esp32doit-devkit-v1\SD/utility/Sd2Card.h:72:30: error: 'SCK_PIN' was not declared in this scope
 uint8_t const  SPI_SCK_PIN = SCK_PIN;
                              ^
*** [.pio\build\esp32doit-devkit-v1\lib28d\SD\SD.cpp.o] Error 1
In file included from .pio\libdeps\esp32doit-devkit-v1\SD\utility\Sd2Card.h:26:0,
                 from .pio\libdeps\esp32doit-devkit-v1\SD\utility\Sd2Card.cpp:26:
.pio\libdeps\esp32doit-devkit-v1\SD\utility\Sd2PinMap.h:371:2: error: #error Architecture or board not supported.
 #error Architecture or board not supported.
  ^
In file included from .pio\libdeps\esp32doit-devkit-v1\SD\utility\Sd2Card.cpp:26:0:
.pio\libdeps\esp32doit-devkit-v1\SD\utility\Sd2Card.h:65:37: error: 'SS_PIN' was not declared in this scope
 uint8_t const  SD_CHIP_SELECT_PIN = SS_PIN;
                                     ^
.pio\libdeps\esp32doit-devkit-v1\SD\utility\Sd2Card.h:68:31: error: 'MOSI_PIN' was not declared in this scope
 uint8_t const  SPI_MOSI_PIN = MOSI_PIN;
                               ^
.pio\libdeps\esp32doit-devkit-v1\SD\utility\Sd2Card.h:70:31: error: 'MISO_PIN' was not declared in this scope
 uint8_t const  SPI_MISO_PIN = MISO_PIN;
                               ^
.pio\libdeps\esp32doit-devkit-v1\SD\utility\Sd2Card.h:72:30: error: 'SCK_PIN' was not declared in this scope
 uint8_t const  SPI_SCK_PIN = SCK_PIN;
                              ^
.pio\libdeps\esp32doit-devkit-v1\SD\utility\Sd2Card.cpp: In member function 'uint8_t Sd2Card::init(uint8_t, uint8_t, int8_t, int8_t, int8_t)':
.pio\libdeps\esp32doit-devkit-v1\SD\utility\Sd2Card.cpp:302:17: error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment
     clkport     = portOutputRegister(digitalPinToPort(clockPin_));
                 ^
.pio\libdeps\esp32doit-devkit-v1\SD\utility\Sd2Card.cpp:304:17: error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment
     mosiport    = portOutputRegister(digitalPinToPort(mosiPin_));
                 ^
.pio\libdeps\esp32doit-devkit-v1\SD\utility\Sd2Card.cpp:306:17: error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment
     misoport    = portInputRegister(digitalPinToPort(misoPin_));
                 ^
.pio\libdeps\esp32doit-devkit-v1\SD\utility\Sd2Card.cpp: In member function 'uint8_t Sd2Card::readData(uint32_t, uint16_t, uint16_t, uint8_t*)':
.pio\libdeps\esp32doit-devkit-v1\SD\utility\Sd2Card.cpp:438:12: warning: unused variable 'n' [-Wunused-variable]
   uint16_t n;
            ^
*** [.pio\build\esp32doit-devkit-v1\lib28d\SD\utility\Sd2Card.cpp.o] Error 1
```

Now I am using 36 pins ESP32 DEVKIT DOIT board. I don't know this is library error or Platform IO error but please guide me what's I am doing wrong.

mobizt, 2021-11-30:
If you are new to PlatformIO and have the question about library installation, please read the [PlatformIO document](https://docs.platformio.org/en/latest/) or ask [here](https://community.platformio.org/).

These libraries are not needed to add to lib_deps as they are in core libraries already.
SPI, FS, SPIFFS, SD

closed 2021-11-30 by mobizt

mobizt, 2021-12-01:
The library in every version before publish, it has been test for both Arduino IDE and PlatformIO in VSCode IDE.

The library itself developed on PlatformIO in VSCode IDE.

mobizt, 2021-12-01:
To make sure you're using the latest ESP32 Platform in PlatformIO, just type this in the console
`pio update`