mobizt/Firebase-ESP-Client #209: BUG ? on Firestore CreateDocuments

quentin-gilli, 2021-12-05:
**To Reproduce**
Steps to reproduce the behavior:
1. Open the exemple Firestore CreateDocuments
2. Only change one thing in the Loop : 
- if (Firebase.ready() && (millis() - dataMillis > _**60000**_ || dataMillis == 0))
- TO --> if (Firebase.ready() && (millis() - dataMillis > **_60*30*1000_** || dataMillis == 0))
3. Launch the program
4. See error on the second query : 

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400da474  PS      : 0x00060e30  A0      : 0x800d6164  A1      : 0x3ffb25a0  
A2      : 0x3ffc3040  A3      : 0x3ffc2e40  A4      : 0x3ffcb118  A5      : 0x3ffb24a0  
A6      : 0x00000000  A7      : 0x3ffb08c0  A8      : 0x800da474  A9      : 0x3ffb2580  
A10     : 0x00000000  A11     : 0x86a1f74d  A12     : 0x86a1f74d  A13     : 0x00000000  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000000d  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x40089754  LEND    : 0x4008975f  LCOUNT  : 0x00000000  

Backtrace:0x400da471:0x3ffb25a00x400d6161:0x3ffb25c0 0x400d61f7:0x3ffb25e0 0x400d72a0:0x3ffb2610 0x400d7311:0x3ffb2630 0x400d7412:0x3ffb26d0 0x400d34b2:0x3ffb2730 0x400e2dc5:0x3ffb2820 

**Expected behavior**
Handle various delay :) 

**IDE and its version:**
 - Arduino
 - Version 1.8.13
 - Lib version 2.7.1

**ESP32 Arduino Core SDK version**
 - Version 2.0.1

mobizt, 2021-12-05:
Post the code, the serial printing message and decoded stack.

quentin-gilli, 2021-12-05:
**The code of your exemple (with no change beside tokens and the one mention above):** 
`
/**
 * Created by K. Suwatchai (Mobizt)
 * 
 * Email: k_suwatchai@hotmail.com
 * 
 * Github: https://github.com/mobizt
 * 
 * Copyright (c) 2021 mobizt
 *
*/

//This example shows how to create a document in a document collection. This operation required Email/password, custom or OAUth2.0 authentication.

#if defined(ESP32)
#include <WiFi.h>
#elif defined(ESP8266)
#include <ESP8266WiFi.h>
#endif
#include <Firebase_ESP_Client.h>

//Provide the token generation process info.
#include <addons/TokenHelper.h>

/* 1. Define the WiFi credentials */
#define WIFI_SSID "..."
#define WIFI_PASSWORD "...."

/* 2. Define the API Key */
#define API_KEY "......"

/* 3. Define the project ID */
#define FIREBASE_PROJECT_ID "...."

/* 4. Define the user Email and password that alreadey registerd or added in your project */
#define USER_EMAIL "...@....com"
#define USER_PASSWORD "...."

//Define Firebase Data object
FirebaseData fbdo;

FirebaseAuth auth;
FirebaseConfig config;

unsigned long dataMillis = 0;
int count = 0;

void setup()
{

    Serial.begin(115200);

    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
    Serial.print("Connecting to Wi-Fi");
    while (WiFi.status() != WL_CONNECTED)
    {
        Serial.print(".");
        delay(300);
    }
    Serial.println();
    Serial.print("Connected with IP: ");
    Serial.println(WiFi.localIP());
    Serial.println();

    Serial.printf("Firebase Client v%s\n\n", FIREBASE_CLIENT_VERSION);

    /* Assign the api key (required) */
    config.api_key = API_KEY;

[... 106 lines cut ...]
load:0x40078000,len:13540
load:0x40080400,len:3604
entry 0x400805f0
Connecting to Wi-Fi............
Connected with IP: 192.168.1.48

Firebase Client v2.7.1

Token info: type = id token, status = on request
Token info: type = id token, status = ready
Create a document... Document already exists: projects/statmaison/databases/(default)/documents/a0/b0/c0/d0

**Decoded stack:**
PC: 0x400da474: FirebaseJsonBase::mSet(char const*, MB_JSON*) at /Users/quentingilli/Documents/Arduino/libraries/Firebase_Arduino_Client_Library_for_ESP8266_and_ESP32/src/json/MB_String.h line 292
EXCVADDR: 0x00000000

Decoding stack results
0x400da471: FirebaseJsonBase::mSet(char const*, MB_JSON*) at /Users/quentingilli/Documents/Arduino/libraries/Firebase_Arduino_Client_Library_for_ESP8266_and_ESP32/src/json/FirebaseJson.cpp line 1045
0x400d6161: NTPClient::getFormattedDate(unsigned long) at /Users/quentingilli/Documents/Arduino/libraries/NTPClient/NTPClient.cpp line 172
0x400d61f7: NTPClient::getFormattedDate(unsigned long) at /Users/quentingilli/Documents/Arduino/libraries/NTPClient/NTPClient.cpp line 179
0x400d72a0: UtilsClass::sdTest(fs::File) at /Users/quentingilli/Documents/Arduino/libraries/Firebase_Arduino_Client_Library_for_ESP8266_and_ESP32/src/Utils.h line 1778
0x400d7311: Firebase_ESP_Client::~Firebase_ESP_Client() at /Users/quentingilli/Documents/Arduino/libraries/Firebase_Arduino_Client_Library_for_ESP8266_and_ESP32/src/common.h line 645
0x400d7412: Firebase_ESP_Client::begin(fb_esp_cfg_t*, fb_esp_auth_signin_provider_t*) at /Users/quentingilli/Documents/Arduino/libraries/Firebase_Arduino_Client_Library_for_ESP8266_and_ESP32/src/./json/MB_String.h line 322
0x400d34b2: EnregistrerDataNRFFirestoreArray(unsigned char) at /Users/quentingilli/Library/Arduino15/packages/esp32/hardware/esp32/2.0.1/cores/esp32/WString.h line 349
0x400e2dc5: __pinMode at /Users/quentingilli/Library/Arduino15/packages/esp32/hardware/esp32/2.0.1/cores/esp32/esp32-hal-gpio.c line 207

mobizt, 2021-12-05:
Check again what I request.

mobizt, 2021-12-05:
Where is printing message?

mobizt, 2021-12-05:
And you should format the code using github markup.

mobizt, 2021-12-05:
The code you post and decoded stack are inconsistent.

closed 2021-12-05 by mobizt

mobizt, 2021-12-05:
Test it already and no error found.