mobizt/Firebase-ESP-Client #218: BUG - Check cloud storage Authentication before uploading

bhupiister, 2021-12-18:
**Describe the bug**
When Authentication to google cloud storage is not checked beforehand and upload is started, it goes through the pain of uploading and fails at last

**To Reproduce**
Steps to reproduce the behavior:
1) Enter false authentication information
2) Start uploading the file
3) It will actually start uploading as seen in the callback function
4) It will finally fail with an authentication error or Panic at Core0

**Expected behavior**
Token info gives information on successful signing, as below. If we can create a check and only then allow upload.

```
Token info: type = OAuth2.0 access token, status = on signing
Token info: type = OAuth2.0 access token, status = on request
Token info: type = OAuth2.0 access token, status = ready
```

**IDE and its version:**
 - PlatformIO

closed 2021-12-18 by mobizt

bhupiister, 2021-12-18:
What is the correct usage then?

mobizt, 2021-12-18:
If you provide the wrong auth credentials, the token generation process will be failed instead of complete due to request rejected by googleapi host.

I can't do anything with your issue.

You should information e.g. library version, sdk version and code that anyone can reproduce or test.