mobizt/Firebase-ESP-Client #226: Token error

Gideon-Jonkers, 2022-01-10:
Hello I'am getting this frustrating token error and when I remove my ntp time sync code the error goes away. I do sync the time before firebase.begin(). This is my token error from the Serial console:

Connected with IP: 192.168.8.26

Firebase Client v2.7.7

Token info: type = id token, status = on request
Token info: type = id token, status = error
Token error: code: -4, message: not connected
Token info: type = id token, status = on request
Token info: type = id token, status = error
Token error: code: -4, message: not connected
Token info: type = id token, status = on request
Token info: type = id token, status = error
Token error: code: -4, message: not connected
Token info: type = id token, status = on request

Does anyone know what I am doing wrong?

mobizt, 2022-01-10:
Should post in [Discussions](https://github.com/mobizt/Firebase-ESP-Client/discussions).

You should test internet accessibility with ping.

closed 2022-01-10 by mobizt

avryux, 2022-10-28:
> Hello I'am getting this frustrating token error and when I remove my ntp time sync code the error goes away. I do sync the time before firebase.begin(). This is my token error from the Serial [...]

Hi man! i have the same error i can't solve it. I'm using NTP too, how could you solve it? Thank u

Gideon-Jonkers, 2022-10-28:
> 
> Hi man! i have the same error i can't solve it. I'm using NTP too, how could you solve it? Thank u

For me it turned out that I had a faulty board. I once put it on a breadboard that was shorted on some pins and when I plugged it in it started the smoke. It still worked but the a few months later trying to implement firebase it was very inconsistent with results. Even different with resets. It sent me on a rollercoaster changing code and I started to get confused. Until I bought a new board everything was solved and the code started to make sense to me again and lined up with the docs. This is what solved it for me. I was using the nodemcu 1.0 I think. It might be a different version.

avryux, 2022-10-28:
i know when this error appears and in my case seems to be when the BL it's on. If i turn on the BL and then signUp to firebase, i get the token error. I tried to do it in the other order and the token error disappear but when i want to get some data in the DB the request failed. :/

mobizt, 2022-10-28:
@avryux 

Additionally, you can't use WiFi and Bluetooth in Arduino simultaneously as they share the same radio signal. https://esp32.com/viewtopic.php?t=6707

You should try NimBLE Bluetooth library because it uses less memory.