mobizt/Firebase-ESP-Client #229: Firebase RTDB

PerumalD, 2022-01-11:
I tried to download the data from rtdb ...It is received successfully. But not write in SD. But successfully write 1kb file but not write large files
Firebase ESp client version :2.7.5
[image]

mobizt, 2022-01-11:
You should follow the issue template when you open the issue.

I can't help without information, code and debug message.

Your code to post here should keep minimal and clear which everyone can test it without third party libraries.

PerumalD, 2022-01-11:
Okay. I will posted.

PerumalD, 2022-01-11:
void loop()
{

  if (Firebase.ready() && !taskCompleted)
  {
    taskCompleted = true;

    //File name must be in 8.3 DOS format (max. 8 bytes file name and 3 bytes file extension)
    Serial.printf("Set file... %s\n", Firebase.RTDB.setFile(&fbdo, mem_storage_type_sd, "test/file/data3", "/file2.txt") ? "ok" : fbdo.errorReason().c_str());

    Serial.printf("Get file... %s\n", Firebase.RTDB.getFile(&fbdo, mem_storage_type_sd, "test/file/data3", "/file3.txt") ? "ok" : fbdo.errorReason().c_str());

    if (fbdo.httpCode() == FIREBASE_ERROR_HTTP_CODE_OK)
    {

#if defined(ESP32)
      Firebase.sdBegin(13, 14, 2, 15); //SS, SCK,MISO, MOSI
#elif defined(ESP8266)
      Firebase.sdBegin(15);  //SS
#endif

      //Readout the downloaded file
      file = DEFAULT_SD_FS.open("/file3.txt", FILE_READ);
      int i = 0;

      while (file.available())
      {
        if (i > 0 && i % 16 == 0)
          Serial.println();

        uint8_t v = file.read();

        if (v < 16) 
          Serial.print("0");

        Serial.print(v, HEX);
        Serial.print(" ");
        i++;
      }

      Serial.println();
      file.close();
    }
  }
}

PerumalD, 2022-01-11:
`void loop()
{

  if (Firebase.ready() && !taskCompleted)
  {
    taskCompleted = true;

    //File name must be in 8.3 DOS format (max. 8 bytes file name and 3 bytes file extension)
    Serial.printf("Set file... %s\n", Firebase.RTDB.setFile(&fbdo, mem_storage_type_sd, "test/file/data3", "/file2.txt") ? "ok" : fbdo.errorReason().c_str());

    Serial.printf("Get file... %s\n", Firebase.RTDB.getFile(&fbdo, mem_storage_type_sd, "test/file/data3", "/file3.txt") ? "ok" : fbdo.errorReason().c_str());

    if (fbdo.httpCode() == FIREBASE_ERROR_HTTP_CODE_OK)
    {

#if defined(ESP32)
      Firebase.sdBegin(13, 14, 2, 15); //SS, SCK,MISO, MOSI
#elif defined(ESP8266)
      Firebase.sdBegin(15);  //SS
#endif

      //Readout the downloaded file
      file = DEFAULT_SD_FS.open("/file3.txt", FILE_READ);
      int i = 0;

      while (file.available())
      {
        if (i > 0 && i % 16 == 0)
          Serial.println();

        uint8_t v = file.read();

        if (v < 16) 
          Serial.print("0");

        Serial.print(v, HEX);
        Serial.print(" ");
        i++;
      }

      Serial.println();
      file.close();
    }
  }
}`

PerumalD, 2022-01-11:
`
[issues.txt](https://github.com/mobizt/Firebase-ESP-Client/files/7845155/issues.txt)
`

PerumalD, 2022-01-11:
```void loop()
{

  if (Firebase.ready() && !taskCompleted)
  {
    taskCompleted = true;

    //File name must be in 8.3 DOS format (max. 8 bytes file name and 3 bytes file extension)
    Serial.printf("Set file... %s\n", Firebase.RTDB.setFile(&fbdo, mem_storage_type_sd, "test/file/data3", "/file2.txt") ? "ok" : fbdo.errorReason().c_str());

    Serial.printf("Get file... %s\n", Firebase.RTDB.getFile(&fbdo, mem_storage_type_sd, "test/file/data3", "/file3.txt") ? "ok" : fbdo.errorReason().c_str());

    if (fbdo.httpCode() == FIREBASE_ERROR_HTTP_CODE_OK)
    {

#if defined(ESP32)
      Firebase.sdBegin(13, 14, 2, 15); //SS, SCK,MISO, MOSI
#elif defined(ESP8266)
      Firebase.sdBegin(15);  //SS
#endif

      //Readout the downloaded file
      file = DEFAULT_SD_FS.open("/file3.txt", FILE_READ);
      int i = 0;

      while (file.available())
      {
        if (i > 0 && i % 16 == 0)
          Serial.println();

        uint8_t v = file.read();

        if (v < 16) 
          Serial.print("0");

        Serial.print(v, HEX);
        Serial.print(" ");
        i++;
      }

      Serial.println();
      file.close();
    }
  }
}```

mobizt, 2022-01-11:
> You should follow the issue template when you open the issue.
> 
> I can't help without information, code and debug message.
> 
> Your code to post here should keep minimal and clear which everyone can test it without third party libraries.

PerumalD, 2022-01-11:
Okay. Downloading the Large files using rtdb. shows this error .
"response payload read timed out due to network issue or too large data size"

mobizt, 2022-01-11:
If you read the issue template, you will see the required information hint.

I don't know which device, sdk version you used.

Read the comments in the example, you will see the guide when you want to work with large file in esp8266.

mobizt, 2022-01-11:
https://github.com/mobizt/Firebase-ESP-Client/blob/6142af83ddc3f8e71ce9a5a2cc80b4445dfe6474/examples/RTDB/File/SD/SD.ino#L88-L91

closed 2022-01-11 by mobizt

PerumalD, 2022-01-11:
Thanks.. But i tried 1kb file successfully write. but 21 kb file not write

mobizt, 2022-01-11:
**fbdo.setBSSLBufferSize(16384, 1024);**

This will consume your mem about 17k+ during operation.

mobizt, 2022-01-11:
You should use the [storage](https://github.com/mobizt/Firebase-ESP-Client/tree/main/examples/Storage) for large file upload/download because in RTDB we need to encrypt as base64 string which the size will go bigger.

PerumalD, 2022-01-11:
Tried in Firebase storage also shows . It seems to be Download File I/0 Error

mobizt, 2022-01-11:
The problem is your SD card or adapter. Check the looseness of cables wiring. The library works with many megabytes files upload and download without problem.

PerumalD, 2022-01-11:
Thanks. Kindly advise them . How to give your file name path in Storage Bucket. It is a getDownloadurl path????

mobizt, 2022-01-11:
Yes, check the comment in the examples when the download url is empty.

PerumalD, 2022-01-11:
How much time take for 1mb file to upload / Download??

mobizt, 2022-01-18:
The [new version](https://github.com/mobizt/Firebase-ESP-Client/releases/tag/v2.8.0) is available with bugs fixed and features added included SdFat support for 5 times faster file read/write, the progress callback for file upload/download.

Please try.

PerumalD, 2022-01-19:
Sure. I will try and feedback to you

PerumalD, 2022-01-19:
Download Failed. File I/o Error

mobizt, 2022-01-19:
You should read the comment in all new examples.

You must choose SD from [here](https://github.com/mobizt/Firebase-ESP-Client/blob/main/src/FirebaseFS.h) and change sd interfaces [here](https://github.com/mobizt/Firebase-ESP-Client/blob/main/src/addons/SDHelper.h).

Fire I/O error is because your SD card is not ready e.g. mounted failed or SPI configuration is not matched to your connection.

PerumalD, 2022-01-19:
Mounting sd card success..

mobizt, 2022-01-19:
SPI configuration is not matched to your connection.

PerumalD, 2022-01-19:
old Version. 2.7.7 it works Perfectly

mobizt, 2022-01-19:
Or the card is failed or cannot access during read/write.

The library always check the media access for every data access.

It's not related to the library version as you think.

mobizt, 2022-01-19:
The old version is loosely check for data access and SD initialization cannot customized.

mobizt, 2022-01-19:
If the amount of data that successfully write or read does not match the byte to write or read, the new library will report the File I/O error while the old version ignore this check and continue to read or write other position.

This means your card is not ready (temporary access failed or SPI connection disconnected) during the continuous read or write.

PerumalD, 2022-01-19:
Thanks to guide.. I will check

PerumalD, 2022-01-19:
Everything is correct configured!!.  Same issue found in ESP-12F Firebase image download

mobizt, 2022-01-19:
I strongly sure that you have problem with the hardware as I have been tested SD thoroughly on ESP32 and ESP8266 devices.

mobizt, 2022-01-19:
You should run the SD card test on your device and check the checksum of data that read and write.

PerumalD, 2022-01-19:
sd card Format in Fat16/Fat32  or else fat

mobizt, 2022-01-19:
FAT or FAT32 is ok except for exFAT and NTFS.

You should try another card.

PerumalD, 2022-01-19:
External ram is needed for nodemcu 12f?

mobizt, 2022-01-19:
No, you should read it, they're options which now available in IDE and sdk v3.0.x.
https://github.com/mobizt/Firebase-ESP-Client#memory-options-for-esp8266

PerumalD, 2022-01-19:
yeah .. Just now also try perfectly work in 2.7.7 .... File size should be 2mb approx. But Updating the version 2.8,.0 not downloading in firebase storage

PerumalD, 2022-01-19:
`void loop()
{
    if (Firebase.ready() && !taskCompleted)
    {
        taskCompleted = true;
    
        Serial.println("\nUpload file...\n");

        //MIME type should be valid to avoid the download problem.
        //The file systems for flash and SD/SDMMC can be changed in FirebaseFS.h.
        if (!Firebase.Storage.upload(&fbdo, STORAGE_BUCKET_ID /* Firebase Storage bucket id */, "/testbmp1.bmp" /* path to local file */, mem_storage_type_sd /* memory storage type, mem_storage_type_flash and mem_storage_type_sd */, "testbmp1.bmp" /* path of remote file stored in the bucket */, "images/bmp" /* mime type */, fcsUploadCallback /* callback function */))
            Serial.println(fbdo.errorReason());
    }
}`

mobizt, 2022-01-19:
Everything was ok and checked in v2.8.0.

mobizt, 2022-01-19:
Do you enable external heap in IDE config even you don't have external SRAM in esp8266?

PerumalD, 2022-01-19:
No.. ...

mobizt, 2022-01-19:
I don't know File I/O error report immediately when trying to open read file or during upload process.

You should check the new examples which the progress callback was added.
Test it and post the debug message here.