mobizt/Firebase-ESP-Client #231: RTDB.getString function returns data with "" quotes

star297, 2022-01-13:
**Just ask something**
If you have questions, use [Discussions](https://github.com/mobizt/Firebase-ESP-Client/discussions) instead. 

**Describe the bug**
Using version 2.7.7
RTDB.getString function returns data with "" quotes added

**To Reproduce**
```
if (Firebase.RTDB.getString(&FBdata, Path))
    {
      Serial.println(FBdata.stringData());

      //  succesfull result adds quotes, e.g. "1234.5678" should be 1234.5678
      //  so function toFloat() no longer works
      
      floatData = (FBdata.stringData().toFloat());
    }
```

**Expected behavior**
Result without quotes as it used to in previous versions but can't remember which version.
I had a look through the examples but couldn't see any mention about that change.

**Screenshots**
none

**IDE and its version:**
 - Arduino
 - Version  1.8.13

**ESP32 Arduino Core SDK version**
 - Version 1.0.6

**Additional context**
Add any other context about the problem here.

mobizt, 2022-01-14:
Thanks I will fix it and update soon.

mobizt, 2022-01-18:
[New version](https://github.com/mobizt/Firebase-ESP-Client/releases/tag/v2.8.0) is available with bugs fixed, please update.

closed 2022-01-18 by mobizt